OpenCPN/OpenCPN #3054: Double define translations are going to wrong translations. (example for german)

BerndCirotzki, 2023-03-07:
I found (for example) two translatoins. One in "opencpn_de_DE.po" and the same in "plugins/dashboard_pi/po/de_DE.po"
with different meanings. 

opencpn_de_DE.po ..
#: src/ais_target_data.cpp:1676
msgid "Port"                                     
msgstr "Hafen"                   ..... this write, when you meen the "harbor", but wrong when you meen "go to port" then it is "Backbord"

plugins/dashboard_pi/po/de_DE.po ...
#: src/dashboard_pi.cpp:1747 src/dashboard_pi.cpp:1970 src/dashboard_pi.cpp:2884
msgid "Port"
msgstr "Bbord"                .... this is right here.

So the Plugin uses the OpenCPN translation (sometimes) ... but this is wrong.
I thing, it is important to use the correct ".po" file for translation by Plugins.

rgleason, 2023-03-07:
Jon Gough has handled this issue in ocpn_draw in a unique way. Perhaps you should ask him what he did?

Hakansv, 2023-03-07:
The "Port" in opencpn..po (ais_target_data.cpp:1676) refers to Port as opposite to Starboard. So the DE translation there would be changed to "Bbord"
This is a common trouble for words like this.
So the general rule for developers is to avoid use of words that can have double meaning in different language. It's not easy.
For the "Port" example " data port" is used and not the single "port". "Harbor" is used and not "Port". 
I searched to code and the word "Port" is used only for the left side of the boat. So far....

Hakansv, 2023-03-07:
> So the Plugin uses the OpenCPN translation (sometimes) ... but this is wrong.
> I thing, it is important to use the correct ".po" file for translation by Plugins.
This is not able to control for the time being. The used translation function can't detach between the core program and plugins in that way. Core OCPN translation seems to get priority. 
Now while beta testing OCPN is the time to identify more such mismatched if any.

BerndCirotzki, 2023-03-07:
So in this case it's very simple it is only a simle translation error.
correct in the OpenCPN translation
to 
#: src/ais_target_data.cpp:1676
msgid "Port"
msgstr "Bbord"

and everything is OK for both cases.

Hakansv, 2023-03-07:
Yes, please update Crowdin

BerndCirotzki, 2023-03-07:
made a pull request.

closed 2023-03-07 by BerndCirotzki