OpenCPN/OpenCPN #3055: RFE: Debug mode needed for new PM protections

rgleason, 2023-03-07:
**Describe the bug - Its not really a bug -Its more of a feature request**
Watchdog on my repos and sean's has diverged. The pypilot alarms on my repos were dated and not working. I removed them. I have tried to make several PR's from Sean's repos to bring the updated Pypilot Alarms into my repos, (with conflicts resolved) but when I copy the dll and pdb files into %localappdata%/opencpn/plugins and run VS there is a very polite message saying that watchdog is not compatible and is being removed.  Therefore there is no way to debug the plugin and find out what is amiss.

Is there a setting in opencpn.ini that we can bypass those protections when using VS debugging ?
**Desktop (please complete the following information if applicable):**
 - OS:Win11
 - O5.7.1

transmitterdan, 2023-03-07:
Rick,

Just create a branch in your fork and edit away the check in the plugin manager.  You might call this branch piDebug.  This is easier and less likely to create bugs with the main branch of code.  To create the branch do this:

```
git checkout master && git pull upstream master && git push && git checkout -b piDebug && git push
```

Git will complain, read what it says and do that.

Now edit away the checks in plugin manager code.

```
git add "all the files you edited" && git commit -m "Edit away plugin manager checks" && git push
```

To bring your local branch up to date with the mainline branch you can do this:

```
git checkout master && git pull upstream master && git push && git checkout piDebug && git merge master && git push
```

Now you have a branch that can debug plugins and it will catch up with the mainline branch every time you do the last bit above.

leamas, 2023-03-07:
Dan's solution is fine.

There are also other work arounds if one reads the code. It depends on the actual dialog displayed when plugins are about to be uninstalled.

If it says something like "The following  plugins have encountered errors during startup" the dialog has a single OK button. However, if the dialog is killed using top-right  X or so nothing is removed.

If it says "The plugin is not compatible with this version of OpenCPN, and will be uninstalled." there is a undocumented environment variable OCPN_KEEP_PLUGINS whcih if set keeps the plugins anyway.

Looking at the code, we should probably respect OCPN_KEEP_PLUGINS ins all cases, this is an oversight.

rgleason, 2023-03-07:
Transmitter Dan, thank you, very helpful and clever, I will try that when I have some time.  I am going to have to find the check in PM. https://github.com/OpenCPN/OpenCPN/pull/3024  This  gets pretty complicated, way over my head.

Also, this is the opencpn.log message that I found
```
19:09:59.895 MESSAGE plugin_loader.cpp:268 Checking plugin candidate: C:\Users\fcgle\AppData\Local\opencpn\plugins\watchdog_pi.dll
19:09:59.895 MESSAGE plugin_loader.cpp:1171 Plugin is compatible: true
19:09:59.895 MESSAGE plugin_loader.cpp:1186 PluginLoader: Loading PlugIn: C:\Users\fcgle\AppData\Local\opencpn\plugins\watchdog_pi.dll
19:09:59.917   ERROR dlmsw.cpp:167 Couldn't find symbol 'create_pi' in a dynamic library (error 127: The specified procedure could not be found.)
19:09:59.917 MESSAGE plugin_loader.cpp:1252    PluginLoader: Cannot load symbol create_pi: C:\Users\fcgle\AppData\Local\opencpn\plugins\watchdog_pi.dll
19:09:59.918 MESSAGE plugin_loader.cpp:248 Loading plugins from dir: C:\Users\fcgle\source\repos\OpenCPN\build\RelWithDebInfo\plugins
19:09:59.918 MESSAGE plugin_loader.cpp:420 PluginLoader searching for PlugIns in location C:\Users\fcgle\source\repos\OpenCPN\build\RelWithDebInfo\plugins
19:09:59.919 MESSAGE plugin_loader.cpp:458 Found 4 candidates
```

rgleason, 2023-03-07:
Such incredible foresight. Thanks.

> If it says something like "The following plugins have encountered errors during startup" the dialog has a single OK button. However, if the dialog is killed using top-right X or so nothing is [...]

rgleason, 2023-03-07:
Just an idea, should this be in the Dev manual perhaps?

leamas, 2023-03-07:
No, this is basically undocumented features we don't want to announce  and thus bind us to that they should exist.

rgleason, 2023-03-07:
ok  got it.  but

After copying dll and pdb files to %localappdata%/opencpn/plugins, starting VS Debugging and 
 hitting the X at the message and it still unloads the dll.

[image]

[image]

rgleason, 2023-03-08:
I have concluded that using the "X" does not remove the plugin dll (which is very good) and much appreciated, however VS also does not run the plugin so that I can debug the bad plugin!

Hakansv, 2023-03-08:
> VS also does not run the plugin so that I can debug the bad plugin!

My method: 
Build the plugin for RelWithDebugInfo by cmake or VS.
Copy the built anyplugin.dll  (and possible the .pdb) to:
(In my case:) C:\Builds\OCPN\OpenCPN\build\RelWithDebInfo\plugins.
BTW: In my case the \plugin is not a folder but a file from initial build. Delete the file and create a folder instead. 
Copy the .dll from the inbound plugins and the one you like observe. (Make a new copy every time you change/rebuild the plugin.)
To brake in the plugin copy the file of interest to VS; OpenCPN. If the brake point is active, filled, when running RelWithDebInfo all is fine. If not make sure VS OpenCPN really sees recent code in "\RelWithDebInfo\plugins"
Clear or "double-talk"?

rgleason, 2023-03-08:
1. Build the plugin for RelWithDebugInfo  with VS or by Cmake
2. Copy the built plugin dll and pdb to  %localappdata%\opencpn\plugins   - The standard directory for plugins which has always worked when debugging plugins.
3. Open VS and navigate to the dll file and open it.
4. Message that this is not part of a project.
5.

leamas, 2023-03-08:
If this ever was an issue it has now morphed to something else. Converting to discussion