OpenCPN/OpenCPN #3056: OCPN log floding    ***GPS Watchdog timeout at ........

Hakansv, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
When no data connection is established the log:
14:04:51.713 MESSAGE comm_bridge.cpp:181    ***GPS Watchdog timeout at Lat:59.3716   Lon: 18.8374
is entered every 5 seconds. This will flood the log when a user is planning or other similar work.

**Describe the solution you'd like**
I think it would be enough to print the log once or twice until next active position is received.
As soon a position is received the log entry would of course be printed again as soon the position watchdog is again overdue.

Thoughts?

leamas, 2023-03-07:
There is [functionality in wxLog](https://docs.wxwidgets.org/3.0/classwx_log.html#ae814f9b9b4c8b4a52e34d7c919f98296) to avoid repeating stuff. However, since we include timestamps each log message is unique, so perhaps it does not work. It depends on where it's hooked up in the logging chain.

We should probably give it a try, if it works it would be a simple fix.

leamas, 2023-03-07:
A simple patch to try: 
```
iff --git a/src/ocpn_app.cpp b/src/ocpn_app.cpp
index 5d6709dc6..76df76800 100644
--- a/src/ocpn_app.cpp
+++ b/src/ocpn_app.cpp
@@ -1209,6 +1209,7 @@ bool MyApp::OnInit() {
            now.FormatISODate().mb_str().data());
   wxLogLevel level = wxLog::GetLogLevel();
   LOG_INFO("Using loglevel %s", OcpnLog::level2str(level).c_str());
+  wxLog::SetRepetitionCounting();
 
   wxString wxver(wxVERSION_STRING);
   wxver.Prepend(_T("wxWidgets version: "));

```

Hakansv, 2023-03-07:
The SetRepetitionCounting is described:
// log the count of repeating messages instead of logging the messages
    // multiple times
But we really do want the same message to be repeated if there has been a valid position that once again disappears.
So I think it have to be coded in comm_bridge?

bdbcat, 2023-03-07:
OK, we can make a simple trigger in comm_bridge..  If position has never been received, post the log message once every (n)  minutes.  When (if?) position is received, revert to once every 5 secs on lost position.
What should be (n)?  Three?

Hakansv, 2023-03-07:
(n) = 60?  It's no need to repeat to the log more often. The log would be used to check for problems earlier received. Instant messaging, if needed, we have on screen. (GPS compass)

Hakansv, 2023-03-07:
The new code works fine, Win10.
I may think 5 seconds after a lost position is quit to much for the log. 
Who is locking there while cruising? Your earlier suggested  three or five minutes would do?

bdbcat, 2023-03-07:
I seem to remember that someone log ago wrote a script to monitor the logfile, and "do something" when watchdog was detected in the logfile.  Might even have been me...  Ring any bells?  That person might want more frequent notifications?
Or maybe my imagination.
I could switch to 3 minutes without anxiety.

Hakansv, 2023-03-08:
No bells here. Could be my age or lack of imagination. :)
One eye on the screen make it clear if no position! Watchdog_pi is prepared for the alarm enthusiasts.
Three minutes is fine.  Very good, thanks!

closed 2023-03-08 by Hakansv