OpenCPN/OpenCPN #3058: Garmin GMR 18 xHD - I can not power it up

pelakuti, 2023-03-07:
Hi, my name is Pelayo, from Spain and I am a cadet on this electronics world.
I have the dream of having my radar on OpenCPN from many years ago and I am right now on the way to try to achieve that goal thanks to the developers of this pluging. I am very grateful. Thank you.
I bought the GMR 18 xHD because of the price and because I dont have any MFC, just the computer, and I read on the wiki that this model would be fitting for this purpose.

I can not power it on and I want to tell you the steeps that I have followed in order to find the mistake:
Right now the antenna is connected to the 24V terminal block bus from my boat. I will take the radar home and connect it directly to a battery to continue the testings here.
I bought the famous Ampherol MRJ 570 but I did not solder the cables on it and I did not attach any switch (I know that on the wiki is written that this is compulsory) because I have seen pictures on the net from other users and they are all different :(
1
[image]

2
[image]

3
[image]

Wich one of the three is the correct? Once I know this for sure I will solder this and attach the switch.
After reading the wiki, in my opinion the connection should me made between this 2 pins:
AMPH
[image]

Am I right? I proceed with the soldering and the switch?
Another question is if I have to solder any aditional cable and if I have to cenect it somwhere else? I know I am in the 2nd scenario (no MFD, directly to PC) but in the picture of the scenrio 1, the ground symbol made me crazy. Do I have to connect the ethernet pin 5 to the battery? Is not just to connect pin 5 from the Amphenol to the "conector shell, chasis ground of the Amphenol" too?
Regarding to the ethernet cable that goes to the computer: does it has to be any kind of special apart from beeing cat 5 minimun? I am using a simple cat6e. Do I have to cut the 5 wire from this cable?

I am desperate, I tought that this would be easy but has become a nightmare to me. I could not power it on, the Status LED is all the time off. In all the forums I have read, I found people with the same issues all of them where not replayed. I have read all the tabs I found on google related to the pin 5 :(
I am praying on my knees for your help and that maybe the origin is just in the non soldering.....
Apologize, is the first time I do this and I have no many idea.
Thank you very much and cheers for your amazing job!

rgleason, 2023-03-07:
Here is some documentation https://opencpn-manuals.github.io/main/radar/0.1/Hardware-installation-(Garmin).html#_suggested_hardware_for_power_control
I haven't used the Garmin or tried this connection...  Comparing the photos it looks to me like you have it right, but you should listen to others more knowledgeable.   I believe you'll get it working so relax and go step by step.

pelakuti, 2023-03-07:
Thank you Rick. That you replied so fast bring me new hope!

leamas, 2023-03-08:
@pelakuti Welcome to OpenCPN!

That said, "issues" here are reserved for bugs, RFEs etc concerning the OpenCPN program and utilities around, it is not a general support forum. For that reason I'm converting this to a Discussion. It should not change anything for you,  but it simplifies life for us while tracking bugs in the software ;)