OpenCPN/OpenCPN #3059: Options -> Connections -> Serial port ->NMEA2000 User comment unreadable

antipole2, 2023-03-08:
**Describe the bug**
When inspecting a serial port using NMEA protocol, the usr comment field is so small as to be unreadable.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: MacOS more than one version
 - 5.7.1 Beta 4

bdbcat, 2023-03-08:
Strange, looks OK to me on MacOS 11.7.1
[image]

antipole2, 2023-03-08:
Now switch to Protocol NMEA2000.  Forgot to mention that bit - sorry.

bdbcat, 2023-03-08:
No change, still looks fine.
What MacOS version?

antipole2, 2023-03-08:
Both on Ventura 13.1 (M2) and on Catalina 10.15.7 (Intel)
Herewith bigger screenshot in case other choices relevant.
[image]

bcn2, 2023-03-12:
Just to confirm on Ventura 13.1 (M1) - same picture
[image]

bdbcat, 2023-03-14:
Corrected.  New build available  now at:
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.7.1.pkg

bdbcat, 2023-03-16:
Close?

closed 2023-03-17 by antipole2