OpenCPN/OpenCPN #3071: Anchor Watch is not in the right click list.

rgleason, 2023-03-10:
**Describe the bug**
Pull very recent master and build and run.

Right click and there is no Anchor watch selection as shown in
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:anchor_watch

If this is not the way to set an anchor watch, how is it done now?

Would like to try the [new Anchor Watch features here ](https://github.com/OpenCPN/OpenCPN/pull/3049/files)

Also I would like to update the documentation.

**Desktop (please complete the following information if applicable):**
 - OS: Win10
 - Version  571

[image]

bdbcat, 2023-03-10:
Rick...
You cannot set an anchor watch on a mark that is greater than a specified distance from ownship.
The value is specified in config file by variable "AnchorWatchMax".
This variable defaults to 1852 NM.
Does this match your test observation?
I would agree that 1852 miles is a somewhat ridiculous number.  Probably should default to something more like 100 NM.  
We do this to avoid excess clutter on context menus.

rgleason, 2023-03-11:
Thanks, I am unfamiliar with this and will make sure the docs show it.  Will check this out further.

rgleason, 2023-03-11:
I've zoomed into the smallest scale  (Scale indicator in lower right is too small to read.) and the anchor watch does not show.
I am going to try to update wxWidgets to 3.2.2.1  ---when did this happen?

bdbcat, 2023-03-11:
Rick...
No relation to wx322.  Been this logic forever.
The waypoint must be "close" to ownship to enable an anchor watch.  Also, there must be a valid GPS position.

rgleason, 2023-03-11:
Dave pointed out that I need to have GPS connected....!

closed 2023-03-11 by rgleason