OpenCPN/OpenCPN #3072: Reversion of Menus Bug

rgleason, 2023-03-10:
I do not know the extend of this reversion, but there needs to be some backtracking here.

[image]

Win 11
OpenCPN 571

bdbcat, 2023-03-10:
Presumably this is a local build.
If so, you need to update the wxWidgets download you are using.  The new download addresses may be found in the buildwin/windeps.bat file, in github master now.
As always, the "official" build is a specified in the various CI scripts.

rgleason, 2023-03-10:
Dave, so I understand, why is this necessary, aren't we using wxWidgets 3.2.1?   It was working properly earlier.

rgleason, 2023-03-10:
I see you've moved to wxWidgets 3.2.2.1?    Should plugins do the same?  What outfall likely from doing that?

bdbcat, 2023-03-11:
As Baikal pointed out, the fix to allow using wx321 "broke" the Dusk and Night color schemes on Windows.  Moving to wx322 allowed us to fix that bug.
The differences from wx321 to wx322 are minor, and ABI compatible.  The only one that affects us is the menu handling on scaled displays for Windows, as we are discussing here.
So there is no real reason to move plugins to wx322, except in the normal course of events as plugins are updated..  Certainly we don't need (and don't want) a wholesale rebuild of all wx32 dependent plugins for this small and invisible change.

rgleason, 2023-03-11:
Lots of moving parts.... Tough for you.   Do I just download wxWidgets 3.2.2.1 expand and put in right place?

bdbcat, 2023-03-11:
Download three files from the links you will find in windeps.bat.  Expand, and copy over the existing files used for your build.

```
  wget -nv %GITHUB_DL%/v3.2.2.1/wxMSW-3.2.2_vc14x_Dev.7z
  7z x -y -o%CACHE_DIR%\wxWidgets-3.2.2 wxMSW-3.2.2_vc14x_Dev.7z
  wget -nv %GITHUB_DL%/v3.2.2.1/wxWidgets-3.2.2.1-headers.7z
  7z x -y -o%CACHE_DIR%\wxWidgets-3.2.2 wxWidgets-3.2.2.1-headers.7z
  wget -nv %GITHUB_DL%/v3.2.2.1/wxMSW-3.2.2_vc14x_ReleaseDLL.7z
  7z x -y -o%CACHE_DIR%\wxWidgets-3.2.2 wxMSW-3.2.2_vc14x_ReleaseDLL.7z

```

rgleason, 2023-03-11:
Extracted to here C:\Users\fcgle\source\repos\OpenCPN\cache\wxWidgets-3.2.2
Should I rename as 3.2.1?  or change the environment ?   ..Then building plugins will require changes...

bdbcat, 2023-03-11:
Perhaps I just change CMake paths in VS for wxWidgets 3.2.2?
Try that first.

rgleason, 2023-03-11:
[image]

rgleason, 2023-03-11:
after clean and build. 

[image]

rgleason, 2023-03-11:
Dynamic link library,  didn't it fully Build?   or do I have to copy the pdb file to somewhere.

bdbcat, 2023-03-11:
You need to copy the new wx3221 runtime DLLs into the VS RelWithDeInfo directory.
Easiest way to do this:  Build a full OCPN setup package.  Install it.  Then copy all the dlls from the installed OCPN package into RelWithDebInfo.

rgleason, 2023-03-11:
OK, when you say "Install it"  I am a little lost here how do I build a full install package?

rgleason, 2023-03-11:
Anydesk?  is on

rgleason, 2023-03-11:
Fixed

closed 2023-03-11 by rgleason