OpenCPN/OpenCPN #3073: Option menu Quality of Data

RooieDirk, 2023-03-13:
The checkbox "Chart Information Objects" in the options menu doesn't work anymore. Switching on/off using the menubar or shortcut "U" does work fine.
Maybe we should remove it from the Options menu and move it to the "Chart Panel Options" menu??
I can make it a pull request if we agree.

Hakansv, 2023-03-13:
I can't see "Chart Information Objects" and m_bShowMeta is used anywhere but that can be I don't understand the code.
But...
Is "Chart Information Objects" and "ENC Data Quality"  really the same?

rgleason, 2023-03-13:
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:charts:vector_display_tab#chart_information_objects

This is Quality of Data
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:charts:vector_display_tab#quality_of_data

rgleason, 2023-03-14:
When in the lower right Hamburger bar, Display category is set to "All"  the Quality of Data is shown. The other 3 Display Categories do not  show Quality of Data  - is this correct?

Using "All Display category,  Toggling "U" then turns Quality of Data on and off.
 
[image]

RooieDirk, 2023-03-14:
Made a pull request.

bcn2, 2023-03-15:
Implemented
[image]

RooieDirk, 2023-03-16:
I know it is there, but I think a text like: "Make chart meta data available" or something similar in better English;) would be better.

bdbcat, 2023-03-17:
Dirk...
Thanks for the suggestion.
However, I think we will stay with what has been used before.  Trying to minimize UI changes at this point.
Next time....

closed 2023-03-17 by RooieDirk