OpenCPN/OpenCPN #3074: Notice: Testing/Improvement of Keyboard Shortcuts for various OS

rgleason, 2023-03-14:
**Describe the bug**
We should test Keyboard Shortcuts in the various OS environments.
and update the documentation as well as confirm operation.

**To Reproduce**
The keyboard shortcuts are here [https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:manual_basic:user_interface:keyboard_shortcuts]() 

Then test the shortcuts in your OS.

For Win11  I've found that the CMD shortcuts don't really have a key anymore.
Perhaps we should change the documentation?

Hakansv, 2023-03-14:
Cmd ?? That's MacOs?  Ctrl on Win
If I search for Ctrl and Win11 there seems to some games (X-box) may occupy it on certain machines.

leamas, 2023-03-15:
What is the reason for  filing this as a bug rather than a Discussion?  It obviously lacks the criterion of a bug.

rgleason, 2023-03-15:
OK make it a discussion please.