OpenCPN/OpenCPN #3077: Building for MacOS m1 (ARM64) should be easier

piotrkundu, 2023-03-14:
**Is your feature request related to a problem? Please describe.**
At the moment I want a first initial build (Xcode project) to generate and compile. I'm passing some of the lib and include paths manually to CMake and some I've updated in CMakeList.txt. I think building for M1/ARM64 should be easier than that.
https://github.com/piotrkundu/OpenCPN/pull/1/files

This is my cmake command at the moment:
`cmake -G Xcode -DZSTD_LIBRARY=/opt/homebrew/Cellar/zstd/1.5.4/lib/libzstd.dylib -DZSTD_INCLUDE_DIR=/opt/homebrew/Cellar/zstd/1.5.4/include -DLibArchive_INCLUDE_DIR=/opt/homebrew/opt/libarchive/lib -DOPENSSL_CRYPTO_LIBRARY=/opt/homebrew/opt/openssl/lib/libcrypto.dylib -DOPENSSL_SSL_LIBRARY=/opt/homebrew/opt/openssl/lib/libssl.dylib -DOPENSSL_INCLUDE_DIR=/opt/homebrew/opt/openssl/include -DMPG123_INCLUDE_DIRS=/opt/homebrew/Cellar/mpg123/1.31.2/include ..`

**Describe the solution you'd like**
Better documentation for ARM64 builds.

**Describe alternatives you've considered**
Maybe add the /opt/homebrew/... paths to the find_library commands for ARM64 on Mac

**Additional context**
Someone familiar with building on OSX should have look at the ARM64 builds for Mac.

nohal, 2023-03-14:
I don't have an ARM Mac at hand, but it seems pretty weird that the libs on your system are not available under `/usr/local` and you need to point cmake directly to the cellar.

piotrkundu, 2023-03-14:
Developing om ARM mac has mostly been a pain so for.... so don't get one just yet.
The reason is that the M1 Mac has Rosetta (Intel emulation) and `/usr/local/ `was used on the intel based macs, so when M1 come, homebrew made a quick hack to support the ARM64 packages and put the in /`opt/homebrew/`. From now on anything you install (on M1/ARM macs) with brew is going to be placed in /`opt/homebrew/`.

Maybe this can be solved by adding `/opt/homebrew/lib` to PATH variable and then maybe Cmake can resolve that, otherwise `/opt/homebrew/ `as a prefix and then use it in HINTS or PATHS:
`set(CMAKE_PREFIX_PATH /opt/homebrew/lib/)`
```
  find_library(
    MPG123 NAMES mpg123 libmpg123
    PATHS ${CMAKE_PREFIX_PATH}/mpg123/lib NO_DEFAULT_PATH
    REQUIRED
  )
```

I don't know, it's a fishing expedition still. You don't need an M1/ARM mac to emulate this, you can just install homebrew in /opt/homebrew/ and then go from there I guess, maybe removing/renaming /usr/local. 

Here is a blogg post I found explaining what homebrew has done on M1: https://earthly.dev/blog/homebrew-on-m1/

PS. Pointing to cellar was just me quick hacking, don't think you will ever need that.

nohal, 2023-03-14:
I'm not sure I'm able or willing to do this to my primary work machine, it must also do some regular work stuff requiring Homebrew to actually behave the normal way it does on Intel...
To properly support building on ARM macOS we need to be able to build on Apple Silicon either on CircleCI or Github Actions, which AFAICT is still not possible (CCI introduced M1 runners for the paid plans only just a few days ago and Github has them in the roadmap for Q3 2023).
I have done pretty much all the preparations required while cross-compiling on x64 hardware, so it should not be hard to finish proper support for M1, but at least for me it has to wait until we have the real thing available one way or the other. And sure it won't happen before 5.8 is out.
From what I read it should mostly really be just about the paths, so if you feel like it a PR is most welcome.

piotrkundu, 2023-03-14:
Yeah, I'm trying to get to the bottom of this, worst case you can remote to my machine if needed. 
[image]
```
  include_directories(/opt/homebrew/include /opt/homebrew/opt/libarchive/include/)
  link_directories(/opt/homebrew/lib /opt/homebrew/opt/libarchive/lib)
  set(CMAKE_OSX_ARCHITECTURES ${CMAKE_HOST_SYSTEM_PROCESSOR})
```
This seems to solve most of the issues with missing include files and libraries, now the problem is that OpenCPN is trying to build for x86_64 and then link to the arm64 libraries, at that's obviously a no-go :(

pierreganty, 2023-04-18:
I am willing to do some testing if that helps.

nohal, 2023-04-18:
It certainly does. I'm investigating some ways to work around the lack of native ARM based builders we could use, can you try if the package at https://dl.cloudsmith.io/public/nohal/opencpn-plugins/raw/names/OpenCPN-darwin-arm64/versions/5.8.0/OpenCPN_5.8.0-12.pkg works on the Apple Silicon machine?

pierreganty, 2023-04-19:
Crashes immediately. Here is the crash report:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenCPN [96756]
Path:                  /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:            org.opencpn
Version:               5.8.0 ()
Code Type:             ARM-64 (Native)
Parent Process:        zsh [65185]
Responsible:           Terminal [633]
User ID:               501

Date/Time:             2023-04-19 13:33:50.4459 +0200
OS Version:            macOS 13.3.1 (22E261)
Report Version:        12
Anonymous UUID:        2F8B483D-4131-7290-22ED-C5F3E2E69F54

Sleep/Wake UUID:       34D98777-A90A-4EA4-B54B-F875E36B8FB9

Time Awake Since Boot: 81000 seconds
Time Since Wake:       2015 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x0000000104f64000
Exception Codes:       0x0000000000000032, 0x0000000104f64000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page

VM Region Info: 0x104f64000 is in 0x104f64000-0x104f8c000;  bytes after start: 0  bytes before end: 163839
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      dyld private memory         104f60000-104f64000    [   16K] ---/--- SM=NUL  
--->  mapped file                 104f64000-104f8c000    [  160K] r--/rwx SM=COW  ...t_id=bc8447ad
      GAP OF 0x62c7c000 BYTES
      Stack Guard                 167c08000-16b40c000    [ 56.0M] ---/rwx SM=NUL  

Thread 0 Crashed:
0   dyld                          	       0x1988f8034 dyld3::MachOFile::hasMachOMagic() const + 0
1   dyld                          	       0x1988ced00 dyld4::SyscallDelegate::mapFileReadOnly(Diagnostics&, char const*, unsigned long*, dyld4::FileID*, bool*, char*) const + 228
2   dyld                          	       0x1988ce668 dyld4::SyscallDelegate::withReadOnlyMappedFile(Diagnostics&, char const*, bool, void (void const*, unsigned long, bool, dyld4::FileID const&, char const*) block_pointer) const + 88
3   dyld                          	       0x1988c769c dyld4::JustInTimeLoader::makeJustInTimeLoaderDisk(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, bool, unsigned int, mach_o::Layout const*) + 208
4   dyld                          	       0x1988be414 invocation function for block in dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 1680
5   dyld                          	       0x1988bd75c dyld4::Loader::forEachResolvedAtPathVar(dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, dyld4::ProcessConfig::PathOverrides::Type, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 776
6   dyld                          	       0x1988aaf54 dyld4::ProcessConfig::PathOverrides::forEachPathVariant(char const*, dyld3::Platform, bool, bool, bool&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) const + 488
7   dyld                          	       0x1988bd294 dyld4::Loader::forEachPath(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&, void (char const*, dyld4::ProcessConfig::PathOverrides::Type, bool&) block_pointer) + 248
8   dyld                          	       0x1988bd9e8 dyld4::Loader::getLoader(Diagnostics&, dyld4::RuntimeState&, char const*, dyld4::Loader::LoadOptions const&) + 588
9   dyld                          	       0x1988c5738 invocation function for block in dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 412
10  dyld                          	       0x1988f87e8 invocation function for block in dyld3::MachOFile::forEachDependentDylib(void (char const*, bool, bool, bool, unsigned int, unsigned int, bool&) block_pointer) const + 148
11  dyld                          	       0x1988a42d8 dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 296
12  dyld                          	       0x1988f8630 dyld3::MachOFile::forEachDependentDylib(void (char const*, bool, bool, bool, unsigned int, unsigned int, bool&) block_pointer) const + 172
13  dyld                          	       0x1988c5480 dyld4::JustInTimeLoader::loadDependents(Diagnostics&, dyld4::RuntimeState&, dyld4::Loader::LoadOptions const&) + 164
14  dyld                          	       0x1988a8a0c dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 1236
15  dyld                          	       0x1988a7e18 start + 1964

Thread 0 crashed with ARM Thread State (64-bit):
    x0: 0x0000000104f64000   x1: 0x0000000000000000   x2: 0x000000016bc056f8   x3: 0x0000000000040002
    x4: 0x0000000000000003   x5: 0x0000000000000000   x6: 0x000000016bc056f8   x7: 0x0000000000000440
    x8: 0x000000016bc05590   x9: 0x000000016bc05598  x10: 0x000000000100000f  x11: 0x0000000000000001
   x12: 0x000000016bc05d22  x13: 0x0000000000000000  x14: 0x000000023973490a  x15: 0x0000000000000000
   x16: 0x000000000000005c  x17: 0x00000001988a5f60  x18: 0x0000000000000000  x19: 0x000000016bc05c20
   x20: 0x0000000000000003  x21: 0x000000016bc056cf  x22: 0x0000000104f64000  x23: 0x000000016bc056f8
   x24: 0x000000016bc056d0  x25: 0x000000016bc056f0  x26: 0x000000016bc05b90  x27: 0x0000000000000000
[... cut ...]
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: USB 10/100/1000LAN, Ethernet, en4
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: Cypress HX3PD Hub
USB Device: USB 10/100/1000LAN
USB Device: Cypress HX3PD Hub
USB Device: Hi Res USB-C AUDIO
USB Device: Keyboard Hub
USB Device: Apple Keyboard
USB Device: Dock Management Controller
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.
```

nohal, 2023-04-19:
Hmmm... That almost seems like unsigned ARM binaries can't run at all.

nohal, 2023-09-28:
After https://github.com/OpenCPN/OpenCPN/pull/3442 we are probably as user-friendly as we can get on macOS, both `x86_64` and `arm64` build out of the box on systems with dependencies from Homebrew and we have a way to build universal binaries with the correct dependencies.

closed 2023-09-28 by nohal