OpenCPN/OpenCPN #3080: Two fingers zoom and right click no longer works in touch screens in 5.7.1+7868+1db87f7d1

sailoog, 2023-03-16:
Raspberry 4 - Bullseye 64bit - 8-inch touch screen from waveshare (a "clone" of the official 7-inch Raspberry screen using the same drivers and connectors).

**opencpn 5.7.1-1+71ffe26 from flatpak**

Everything works OK out of the box in a fresh installation. 2 fingers zoom in/out, right click with a long tap and the main menu is reacting to clicks from the beginning. Interestingly there is no difference in behavior when the setting "Enable Touchscreen interface" is disabled which is good because in the following case (PPA) it is impossible to open the options window on a system without a keyboard or mouse because the main menu does not react to touch screen clicks until that setting is enabled.

**opencpn 5.7.1+7868+1db87f7d1 from PPA**

In the previous 5.7.1+7816+7abc448d4 everything was working more or less as in the flatpak version but in this new iteration touchscreen features no longer work. Before enabling "Enable Touchscreen interface" you can not open the options window and you are forced to plug in a mouse. After enabling that setting the main menu reacts to the clicks from the touch screen but zoom in/out with 2 fingers and right click with long tap do not work.

Let me know if you need more details.

bdbcat, 2023-03-21:
Sailoog....
1. Background:
The two-finger zoom and right click emulation are provided by system drivers, and are not a part of OpenCPN for GTK.  So, the behavior of long-tap is not at all controlled by OpenCPN.  OCPN simply receives an emulated right-click, as though a mouse were connected.  Similarly, two-finger zoom emulates a mouse wheel zoom activity, and OCPN has no notion that the action comes from a touch-screen, or a real mouse.

2. OCPN current master build supports my touch screen for button clicks, and chart movement.  It does not do two-finger zoom or long-tap operation, but that is because I do not have the correct drivers for it.

3. I built the earlier commit referenced, and there is no change in behavior.

4.  Interestingly, latest flatpak build does support two-finger and long-tap. It must be true that the flatpak build has the correct touch drivers "built-in".
5.  Any references/tutorials on installing drivers for "official" 7" RPI display on latest Raspberry Pi OS would be appreciated.

Thoughts?

sailoog, 2023-03-21:
Dave,

That was my first thought, but the flatpak version made me change my mind. I made 5.7.1+7816+7abc448d4 work at some point, so I will try to go back on my steps configuring the system.

About installing native drivers, the only thing I found is adding this to /boot/config.txt:

dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch

but it does not work for me (RPI4 - bullseye 64bit).

still investigating...

sailoog, 2023-03-21:
Installed the old 5.7.1+7816+7abc448d4 but right click and 2-fingers zoom do not work. That should confirm that this is not an opencpn issue but a system issue. 

I would keep this issue open until a solution is found for the record but feel free to close it if you need to move forward.

rgleason, 2023-03-21:
@mgrouch could help here.

mgrouch, 2023-03-21:
You need wxWidgets 3.2 and my patch (or something derived from it) to OpenCPN. Default bullseye is wxWidgets3.0.  So it doesn’t work in bullseye. 

It does work fine in bullseye based BBN Marine OS for pi4 (but it is solved there on OS packaged level). Try

https://bareboat-necessities.github.io/

mgrouch, 2023-03-21:
You could create .deb packages of wxWidgets 3.2 for bullseye on CircleCi. I’ve done it before and it’s quite easy. You could host them somewhere (release directory on github can be made .deb repository)

Then you could compile and distribute OpenCPN with WxWidgets 3.2 on bullseye too. 
You need to add my patch for handling touch gestures (they are not mouse events) into OpenCPN source code.

mgrouch, 2023-03-21:
The patch was there 
https://github.com/OpenCPN/OpenCPN/issues/2057

mgrouch, 2023-03-21:
There were some commits for it it seems. 
https://github.com/OpenCPN/OpenCPN/pull/2333

sailoog, 2023-03-21:
Maintaining a fork of opencpn or wxWidgets will never be an option for openplotter. The intention is not to collect users, we want any user to be able to install opencpn on a Raspberry. There is a plan for opencpn and wxWidgets 3.2 and openplotter will follow it.

mgrouch, 2023-03-21:
> Maintaining a fork of opencpn or wxWidgets will never be an option for openplotter. The intention is not to collect users, we want any user to be able to install opencpn on a Raspberry. There is a plan for opencpn and wxWidgets 3.2 and openplotter will follow it.

What fork?

sailoog, 2023-03-21:
You know what I mean, compiling and installing unofficial deb packages.

mgrouch, 2023-03-21:
> You know what I mean, compiling and installing unofficial deb packages.

I do not know. Is Openplotter official deb package?

I was talking about where the issue is. And where it is in the code.

sailoog, 2023-03-21:
> You could create .deb packages of wxWidgets 3.2 for bullseye on CircleCi. I’ve done it before and it’s quite easy. You could host them somewhere (release directory on github can be made .deb [...]

mgrouch, 2023-03-21:
> You could create .deb packages of wxWidgets 3.2 for bullseye on CircleCi. I’ve done it before and it’s quite easy. You could host them somewhere (release directory on github can be made .deb [...]

@sailoog 
It was addressed to all OpenCPN developers to consider packaging it that way. it’s OpenCPN issue chat here.

mgrouch, 2023-03-21:
Bullseye is still not having WxWidgets 3.2. (The ones with touch gestures support on Linux)

leamas, 2023-03-21:
> Bullseye is still not having WxWidgets 3.2

And it will never have it. WxWidgets 3.2 comes with bookworm, to be released probably this summer (currently in beta freeze).

sailoog, 2023-03-24:
For the record.

Until we found a solution at system level we can absolutely recommend opencpn from flatpak for touchscreens. 2-fingers zoom and right click are working well there. Finally the fading scrolls can also be disabled in opencpn flatpak (thanks leamas) and just now I have also managed to modify the size of the scrolls. Check out this monstrous scroll perfect for sailor fingers:

[image]

You can do this in Debian and derivatives adding this file to disable fading scrolls:

```
~/.var/app/org.opencpn.OpenCPN/config/gtk-3.0/settings.ini

[Settings]
gtk-overlay-scrolling=false
```
and this one for scroll size:
```

~/.var/app/org.opencpn.OpenCPN/config/gtk-3.0/gtk.css

scrollbar slider {
    /* Size of the slider */
    min-width: 20px;
    min-height: 20px;
    border-radius: 22px;

    /* Padding around the slider */
    border: 5px solid transparent; 
}
```
to do this in native opencpn (no flatpak), this is the right path for both files:

`~/.config/gtk-3.0/`

Always reboot after any modification.

leamas, 2023-03-24:
Since touch screens basically requires wxWidgets 3.2 the overall situation seems to be:
  - It works for Flatpak (which uses wx 3.2)
  - It does not work on native Bullseye which uses wx 3.0. 
  - Native packages will stay this way until we have a bookworm version of OpenCPN which uses wx3.2 (or a custom Bullseye backport of wx 3.2)

Given all this, should bug be kept open?

Hakansv, 2023-03-24:
I've tested adding to ~/.config/gtk-3.0/ for my homebrewed on Bullseye. Works very fine!
Thanks

sailoog, 2023-03-24:
> 
> Given all this, should bug be kept open?

not for me, closing, reopen if necessary.

closed 2023-03-24 by sailoog

bdbcat, 2023-03-24:
"Since touch screens basically requires wxWidgets 3.2"
Not exactly.  Basic touch screen support exists in wx30, provided by a system layer driver the captures touch events, and generates equivalent mouse actions to inject into the input stream.  An app like OCPN is unaware of this process, and in general has a had time detecting that a touch screen is in use.
The basic problem discussed in this thread is this:  this system level touch driver driver for Rpi4/64/Bulleye does not seem to exist. or does not work yet.  Lots of Google juice on this....
I am happy with the flatpak recommendation for this particular hardware use case.