OpenCPN/OpenCPN #3081: DualCanvas ENC settings not well seperated

RooieDirk, 2023-03-16:
If in dual canvas mode with 2 ENC charts the chartcanvas settings are changed in one canvas, it is also used for the second canvas. (At least as soon as the second canvas gets an update. See screenmovie. Starting with both canvasses in 'Base' mode and then change one to all.

https://user-images.githubusercontent.com/2668258/225616431-487c824f-c382-42a3-89d5-8352773998b5.mp4

Hakansv, 2023-03-16:
For info I can't reproduce this neither on my Win10 nor the RPi400. Both canvases can be changed without interfering the other.
I use o-charts on both though.

RooieDirk, 2023-03-16:
The O-charts plugin is indeed fine. I was using plain S57 ENC charts.

bdbcat, 2023-03-18:
Dirk...
Neither Hakan not I can reproduce this. Any more info, or test with most recent source build?
Thanks

RooieDirk, 2023-03-18:
Dave, It only happens with plain ENC charts, the O-charts works as supposed. (Hakan did use wit O-charts)
I did try again, now with some US ENC charts to make sure it is not a faulty chartset I am using. 
The latest master code (10 minute ago) As in previous screen recordings.
With the latest flatpack (test #4)  same as in recording
With the 5.6.2 flatpack things are working normal.
As I get the same effect in flatpack as well, I would think it is not my computer setup.

I'm running Ubuntu 22.04.1 LTS

Hakansv, 2023-03-18:
For the story..
O master at commit: "Correct toggle interference on multi-canvas ...."
On Win10 I'm now using plain ENC covering DK and SE. No interference between the two canvas when toggle anchorage, lights, soundings and more.

bdbcat, 2023-03-19:
Indeed, problem exists only on plain S57 ENCs.  Root cause is error in S52 library state management.
Corrected now, I think.
Please verify.

RooieDirk, 2023-03-19:
Seems indded to be fixed now Thanks

closed 2023-03-19 by RooieDirk