OpenCPN/OpenCPN #3082: Missing vertical scroll in several tabs of options window (O571)

sailoog, 2023-03-16:
Raspberry 4 - Bullseye 64bit - 8-inch screen 800x480

This is the resolution of the official monitor for Raspberry and a very good option for the boat due to its low consumption. In lower resolutions opencpn is very difficult to handle so 800x480 is the minimum resolution that we recommend in OpenPlotter.

There are some tabs in the options window where scrolling is missing and some settings remain hidden and inaccessible even for larger resolutions:

- Charts - Vector Chart display: missing scroll.
- Charts - Chart Groups: missing scroll.
- Charts - o-charts: There is a vertical scroll but once the charts are selected, the "install" button is never shown, even scrolling to the end, making installation impossible.

[image]

bdbcat, 2023-03-18:
Sailoog...
Will you attach/post the logfile for this particular configuration?
I have tested using "official" 7" screen, RPi3/ARMHF, Buster.  Local build of O571 today.
Result:  All Options panels have required scroll bars.
Thanks

Hakansv, 2023-03-19:
The scroll bar is only shown if the dialog height is not enough to cover it all.
What I can see on my RPi is the right scroll bar is not shown if the pointer is outside the dialog window. It's only visible after some pointer movements inside the dialog. I can agree this is sometimes confusing. Is that a wx 3.0 "feature" and not able to control by OCPN?

sailoog, 2023-03-19:
Ok, I'll try again on Monday, but it's weird because I really did a lot of testing. I forgot to say that the options window is maximized.

bdbcat, 2023-03-19:
I reiterate:
Everything working fine on RPI/3 Buster.

After some fiddling, I have migrated my 7" display to RPi/4, Bullseye/64.  To make it work sensibly, one must disable both of the internal HDMI ports using the RPi Preferences->ScreenConfiguration.

Hakan: "What I can see on my RPi is the right scroll bar is not shown if the pointer is outside the dialog window. It's only visible after some pointer movements inside the dialog"

That is true in my arrangement, as well.  The scroll bar "fades away" after no movement in some seconds.  Confusing.  I suspect this is some interaction between wxWidgets and GTK/Window manager.  Not sure if this behaviour is configurable or not.
Further, the scroll bar does NOT appear on some options tabs at all, so this is a clear problem.

I will investigate further.

bdbcat, 2023-03-19:
First discovery:
Fading scrollbars are a GTK3 innovation.  Not universally loved at all.
Simple solution (for testing):
`$export GTK_OVERLAY_SCROLLING=0
`
There are other, more persistent, ways to configure this action.

```
$ cat ~/.config/gtk-3.0/settings.ini

[Settings]
gtk-overlay-scrolling=false

```
Not sure what is the best approach for OpenPlotter...

More as I know it.

leamas, 2023-03-19:
Just a drive-by comment:

Seems that the cli command `gsettings set org.gnome.desktop.interface overlay-scrolling false` should make the trick since gnome on bullseye is 3.38 (the command is only available on gnome 3.34+).  Not tested.

A cli command like this should be fairly simple to integrate in any installation procedure.

See : https://www.linuxuprising.com/2019/09/how-to-disable-gnomes-overlay.html

bdbcat, 2023-03-20:
Status...
Some refactoring of Options dialog required.  In process. a few more hours.  Will be finished tomorrow.

sailoog, 2023-03-20:
Tested on RPi4 - bullseye 32bit and 64bit - opencpn 5.6.2 and 5.7.1 from PPA (focal). Missing scroll in all the tests in the related tabs in the first post.

sailoog, 2023-03-21:
**Missing scrolls**

After testing opencpn - 5.7.1+7888+77a4b15ad~ubuntu20.04.1:

-  Charts - Chart Files: OK
-  Charts - Vector Chart display: OK
-  Charts - Chart Groups: OK
-  Charts - o-charts: OK
-  Dashboard preferences: fixed window size and no scroll (hidden OK/Cancel buttons).
-  GRIB preferences: fixed window size and no scroll (hidden some settings and OK/Cancel buttons).
-  chartDownloader preferences: fixed window size and no scroll (hidden OK/Cancel buttons).
-  o-charts preferences: fixed window size and no scroll (hidden some settings and OK/Cancel buttons).

**Scroll overlay**

This worked in OpenPlotter 3/Raspberry bullseye:

```
$ cat ~/.config/gtk-3.0/settings.ini

[Settings]
gtk-overlay-scrolling=false
```
Yes, we can take care of that adding a setting in openplotter-settings (Raspberry settings tab)

bdbcat, 2023-03-21:
So, all good on this now?

Hakansv, 2023-03-21:
Scroll overlay..
Can this affect Androids in the same way? There will a "fading out" scroll bar create some not wanted experience as well.

bdbcat, 2023-03-21:
I am currently working on adding scrollbars to Dashboard preferences.  Others will follow.

leamas, 2023-03-23:
> This worked in OpenPlotter 3/Raspberry bullseye (although not working in flatpak):

Flatpak's config directory is in  _~/.var/app/org.opencpn.OpenCPN/config/gtk-3.0/_

bdbcat, 2023-03-24:
Remaining problems with Dashboard Preferences width now corrected in github, and PPA for linux.

sailoog, 2023-03-24:
> > This worked in OpenPlotter 3/Raspberry bullseye (although not working in flatpak):
> 
> Flatpak's config directory is in _~/.var/app/org.opencpn.OpenCPN/config/gtk-3.0/_

That worked too. Thanks!

sailoog, 2023-03-24:
> Remaining problems with Dashboard Preferences width now corrected in github, and PPA for linux.

It works perfect now in 5.7.1+7901+7ff014ee2~ubuntu20.04.1

Only the o-charts preferences left. Thanks!

bdbcat, 2023-03-25:
o-charts_pi updated to v1.0.34.0 for this platform.  In plugin master catalog now.

sailoog, 2023-03-25:
Everything working now, thanks!!
Closing...

closed 2023-03-25 by sailoog

RoDeP0, 2023-03-30:
Regarding the O-charts plug-in: the scroll bars are back, but the update button is still not available, which makes in-app updating impossible. Have to add that I updated to o-charts_pi v 1.0.34.0 on OpenCPN 5.6.2, because it's a production system. So it may work on OpenCPN 5.7.1+7901+7ff014ee2~ubuntu20.04.1. Raspberry Pi 4 model B 4 GB with official 7" DSI touch screen.