OpenCPN/OpenCPN #3086: Grib: Artifact at minimal zoom

nohal, 2023-03-16:
At maximum zoom-out a distorted second copy of the grib heat map appears when grib area gets close to left/right edge of the screen.

**To Reproduce**

0. OpenGL enabled
1. Load a grib
2. Have the heat map enabled
3. Zoom all the way out
4. Pan towards left or right until the artifact appears

[image]

nohal, 2023-03-24:
Better with current master, but the artifact still can be summoned by panning the grib covered area off the screen at ver small scales,

bdbcat, 2023-03-25:
Please try current master.  I took a different approach, looks OK on the tests I have tried.
If trouble is seen, please include link to GRIB file.

nohal, 2023-03-25:
Fixed.