OpenCPN/OpenCPN #3092: Send to peer dialog - Cancel while scan - Busy spinner for ever

Hakansv, 2023-03-19:
**Describe the bug**
A user decide to abort a Send to peer action while the scan process is ongoing.
The Scan to peer dialog is then closed but the mouse cursor is left at "Busy spinner" forever until O is closed.

**To Reproduce**
Start a Send To Peer function.
While the scan process is performed and the gauge is growing click the Cancel button.
The dialog is closed but the busy spinner is still present until O is closed

**Expected behavior**
Reset the mouse cursor when the dialog is closed or omit the Cancel button during scan.

**Desktop (please complete the following information if applicable):**
 - Win10 and RPi Debian

**Screenshots**
Click the Cancel button here:
[image]

closed 2023-03-19 by nohal

Hakansv, 2023-03-19:
Bug fix in 5 minutes! 
Must be some kind of record.
Thanks

nohal, 2023-03-19:
If just all of them would be this easy :)

commit 71d64d2e (2023-03-19): End the busy spinner on send to peer dialog cancelation. Closes #3092