OpenCPN/OpenCPN #3093: API-118 ... PlugIn_Route_Ex - Unable to search for a route with non-existent GetSingleRouteEx

Rasbats, 2023-03-20:
`GetSingleWaypointEx` works perfectly for finding a waypoint from an array of waypoints.
There is no code `GetSingleRouteEx` to extract a route from an array of _Ex routes.

**Expected behavior**
A function to mirror that of GetSingleWaypointEx:
```
extern DECL_EXP bool GetSingleWaypointEx(wxString GUID,
                                         PlugIn_Waypoint_Ex *pwaypoint);

```
**Additional context**
Having this function would avoid finding a route from a waypoint on that route, which may way exist for two or more routes.

antipole2, 2023-03-20:
Not sure what you need that is not available through
`GetRouteEx_Plugin(GUID)`
That is how I have implemented the JavaScript
`route = OCPNgetRoute(GUID);`

Rasbats, 2023-03-20:
Thanks. My declaration was the problem. Closing.

closed 2023-03-20 by Rasbats