OpenCPN/OpenCPN #3098: AIS shortcut W preventing some plugin typing

antipole2, 2023-03-20:
**Describe the bug**
In certain plugin text fields, typing w or W gets treated as an AIS `Show CPA Warnings` shortcut instead of being entered in the text field.

**To Reproduce**
Steps to reproduce the behaviour:
1. Activate the JavaScript plugin
2. In either the script window or the results window type upper or lower case W.  The character does not appear.  Instead, the AIS menu item flashes, indicating it has been handled as a shortcut.

**Expected behaviour**
The character should appear in the text field as it did under OCPN5.6.2 before the W shortcut was added to the AIS menu.

**Desktop (please complete the following information if applicable):**
 - OS: macOS
 - Versions: This is the case for both Catalina and Ventura

**Additional context**
The problem does not seem to exist for other plugins I have tested.  Also in the JavaScript plugin, if you open the Tools window, you can type w in, say, the name of a new Console.
The two text fields that have shown the problem are both of type `wxStyledTextCntl`.

The problem does not occur with the characters + or - which work as scaling shortcuts when the focus is on the OCPN canvas but appear in a text field if the focus is on its window.  It seems like OCPN is grabbing the w character as an AIS shortcut regardless of whether OCPN has the focus.

NB w is a useful character when typing waypoint!

Hakansv, 2023-03-20:
What happens with all other key characters used in O like O, Q, U, A, S, T, L? Is there a difference to W?
I can use "W" and the other in O's dialogs like Route &Mark manager and on other plugins like Watchdog and more.

antipole2, 2023-03-20:
No problems with  O, Q, U, A, S, T, L, W nor with o, q, u, a, s, t, l
Only with lowercase w and, as far as I know, only when the text field is `wxStyledTextCntl`

The oddity is that there is no issue under OCPN 5.6.2.
So we could point the finger at wx3.2 but it seems a bit too much of a coincidence that the only character to give this problem is 'w' and 'W' has been added as a shortcut to the AIS menu.

Hakansv, 2023-03-20:
I see you're on a Mac.
It may be I unintentional made a mistake with this. macOs seems to handle this in another way. I'll have look.

bdbcat, 2023-03-21:
Corrected by Hakansv PR.
Thanks

bdbcat, 2023-03-21:
Close?

antipole2, 2023-03-21:
Apologies for my ignorance - but where do I find the latest builds so I can confirm this for myself?

Hakansv, 2023-03-21:
Try [here](https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/) I suppose the "OpenCPN_5.7.1.pkg " will fit you. My Mac knowledge is embarrassing low.