OpenCPN/OpenCPN #3103: PluginManager show false installed plugin version after a Import Tarball.

Hakansv, 2023-03-22:
**Describe the bug**
After a Tarball import the plugin version number is not correct.
I think PM use last update version from catalog master instead of actual plugin version.
PM can later on also ask to "Downgrade" even if it's a false message.
More discussion [here on zulip](https://opencpn.zulipchat.com/#narrow/stream/332168-Master---5.2E8.2E0--.28was.3A-comms.29/topic/Test.20issues/near/339020574)
**To Reproduce**
Steps to reproduce the behavior:
1. Import and install a tarball.
2. Show preferences for a plugin where the actual version is shown e.g. o-charts 
3. Compare announced versions.

**Expected behavior**
PM would show installed version rather than some historical catalog update.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS:The example shot is from a RPi400
 - Present Git master

rgleason, 2023-03-22:
> PM can later on also ask to "Downgrade" even if it's a false message.
I have noticed this often.

leamas, 2023-03-22:
This has been discussed in depth on [zulip](https://leamas.github.io/zulip-archive/stream/332168-Master---5.2E8.2E0--.28was.3A-comms.29/topic/Unloadable.20old.20plugins.20blues.html#327954149)

The short story is that we made a fix for the unloadable  plugins which we inevitably encounter when we do an ABI break. This was the easy one.

However, the still standing issue is illustrated in what happens when you import a plugin which is not part of the catalog. After a catalog update this plugin will not be visible in the GUI since it is not part of the catalog, even though it's actually installed. In more general terms, the current design often means that the catalog data is not synced with imported plugins.

I and @bdbcat  have agreed that the correct solution to this mess would be to never overwrite the plugin metadata for an imported plugin when the catalog is updated.  It should be left in place until the imported plugin is uninstalled. 

However, this is a big change and I don't think this is the time for a major redesign of the catalog handling. That is, I think the right handling of this and related bugs is to live with them in 5.8 and fix it in next cycle.  

TL; DR: The bug is real, but I think it should be deferred.

Hakansv, 2023-03-22:
The simple Q would be; How can o-charts preferences read the correct version if PM can't?

Hakansv, 2023-03-23:
Thanks for comments and I'll not force work on this now. No problem.
Just a Q if a quick answer is available.. Where does then the preferences dialog gets the version designation if not from metadata?

leamas, 2023-03-24:
Assigning to next release. Solving means handling the general problem of keeping imported metadata consistent, in particular when the catalog is updated.

leamas, 2023-05-16:
@Hakansv  I have #3103 in the oven. If you have some free cycles, something I certainly doubt, but still: Could you possibly check if it solves the issues you describe here?

Hakansv, 2023-05-16:
I'll try tonight or tomorrow.

Hakansv, 2023-05-16:
@leamas 
I've tested PR #3258 on Win10
I imported a home brewed Tarball of radar_pi. 
Version before the patch was false -1.0.0 and after build of 3258 correct 5.5.0
Very good!

Before:
[image]
After:
[image]

Hakansv, 2023-05-16:
@leamas 
Looking in "C:\ProgramData\opencpn\plugins\install_data\radar.version
it says: 5.5.1-alpha+2304221749.0d91def

leamas, 2023-05-16:
Yes, this is actually not good and not how it works for me. I'll sleep on it and see if I can reproduce.

Just a quick check: Does anything happen if you close and re-open the the options window?

leamas, 2023-05-16:
Hm.. I think I know what the problem might be. Have you got a link to your plugin sources?

Hakansv, 2023-05-16:
I tested on radar_pi and that tarball was built only locally. I use no other Git than https://github.com/opencpn-radar-pi/radar_pi

leamas, 2023-05-16:
Any changes  to Plugin.cmake  in your build?

Hakansv, 2023-05-16:
Well I haven't updated from Git master so still the same as when that Tarball was  built:
set(PKG_NAME radar_pi)
set(PKG_VERSION 5.5.1)
set(PKG_PRERELEASE "-alpha")

leamas, 2023-05-16:
The interestin fact is that when I rebuild from radar_pi @5729cec5 everything works fine for me: Both fields reports the version as of below
[image]

leamas, 2023-05-16:
So, again, let's hunt some differences. The first is that I'm on linux and you on windows. However, I have worked a lot with this code and I'm not aware of any platform specific code paths w r t this.

Next question: Are you on 5729cec5 for radar_pi?

And, given some late experiences: Stil the same  of you rebuild from scratch?

leamas, 2023-05-16:
good night. Tomorrow is a new day ;)

Hakansv, 2023-05-17:
> Next question: Are you on 5729cec5 for radar_pi?

No the used tarball is from mid April. If you want to test it's [here](https://www.dropbox.com/s/nm9ubgp1zzroby8/Radar-5.5.1-alpha%2B2304221749.0d91def_msvc-wx32-10.0.19044-win32.tar.gz?dl=1).
I could make a fresh one from Git master if you think that's a difference. To be able to detach that and the one from master catalog I've to make it another version.

leamas, 2023-05-17:
Works fine for me also on windows using your radar_pi link above:

[image]

Hakansv, 2023-05-17:
I made a rebuild all from VS and a package as well. (No new dependencies though.)
With that version of O the import function failed. No "Import succeed" message and the  install fail due to "bad url".
If it's my build to be questioned you may have a Cloudsmith package to test?

leamas, 2023-05-17:
hm... this does not feel stable. for sure. But I have anyway found a problem which is ridiculously hard to fix.

Of course, we could find som cloudsmith tarballs to test. Please stay tuned.

leamas, 2023-05-17:
I have solved the problem with **[Imported]** not being shown in the first view after importing. No questions about why it took so long, please...-

For me this works out of the box also on windows. I use the same radar plugin tarball as you, the dropbox link [above](https://github.com/OpenCPN/OpenCPN/issues/3103#issuecomment-1550890115). Given or common experiences, a build problem seems likely.

It would be a lot of work for me to get started with Visual Studio, and I'm not really interested in it anyway working mostly on Linux. Could you possible try to do the same as me i. e., build a package and install it according to the manual?

leamas, 2023-05-18:
> There was no response message about a successful import but the version number for radar_pi instantly changed although not yet installed-?? - Pict 1

No, this is not what is happening. New code installs it directly without the actually a little strange "old" way of only installing into the catalog.

> Now the version is v5.5.0 although imported 5.5.1-alpha is installed. Pict 2

And then again, what you do when you install from the catalog is that you install from the catalog. It does what it should, but it seems to confuse you.

I actually think the new behaviour is much more intuitive. Being a  Linux guy I'm not that fond of dialogs saying that things which I try do do actually worked, it should not be a surprise, should it? But perhaps we should have a "Installation complete" dialog at least as a  transition state.  Do you think this would make it more clear to you?

leamas, 2023-05-18:
However, updating the catalog does **not** do the right thing. More bugs to fix...

Hakansv, 2023-05-18:
OK. I'll be more or less off due to a meeting here at Läsö with JRSK: "SE: Jorden runt seglarklubben" -> Circumnavigation association.
Including some OCPN support....

leamas, 2023-05-19:
This is a can of worms, I need to rewrite large parts. Expect some delays...

leamas, 2023-05-23:
So, I'm back with a new version. This is a heavy rewrite, when I digged deeper into plugin_loader and friends I found some really bad parts. Should be better now, plugin_loader might actually be fine.

Of course, I'm happy if you have some time...

Hakansv, 2023-05-25:
Alec..
I have build PR 3258.
These are what's seen on my Win10.
Håkan

First view wo actions:
PM says v5.3.4
Actually installed is 5.5.1-alpha...
[image]
Then I updated master catalog and installed 5.5.0:
PM says v5.3.4
Actually installed is v5.5.0
[image]
Then import 5.5.1-alpha...
I do appreciate the message:
[image]
Now PM says: 5.5.1-alpha.... (Imported)
Actually installed is v5.5.0
[image]
Then I installed the imported 5.5.1-alpha
PM says: v5.3.4  (From where??)
Actually installed: 5.5.1-alpha....
[image]

leamas, 2023-05-25:
you are a good test case, thanks. Will look into it.

leamas, 2023-05-25:
Back to terminal after kitchen management.

This was a nasty bug, not that hard to explain: I need to match the loaded plugin with correct metadata file. I only used the name in this match, so the used metadata file was a random pick among those with correct name. Now fixed and works for me (tm) on windows using Radar...

Hakansv, 2023-05-25:
OK. good.
I'll try to check your patch tomorrow. Now soon bed and a lot others tomorrow but I'll do my best.

leamas, 2023-05-25:
> Now soon bed

indeed. Good night.

I do appreciate all your efforts, really, and I'm just happy when you have some time for this. Of course we all have too much to do...

Hakansv, 2023-05-28:
Alec. Sorry for the delayed response.
There's a difference but not ready yet..
Actions story:

1. First visit. No actions:
PM and actual are synchronized.
[image]
2. Update master catalog and install 5.5.0
PM and actual are synchronized.
[image]
3. Import a tarball
After import and no further actions.
PM says 5.5.1-alpha....imported
Actual says v5.5.0
Note: No buttons for Reinstall nor Uninstall 
[image]
4. After O restart
PM and actual are synchronized.
But still no buttons and the right status icon is grey!
And to disable has no function. It's still enabled after Apply/OK. 
No difference after O restart.
[image]

I had to manually delete radar_pi.dll and install from master catalog to get the buttons back.
I also deleted all plugin cache for radar_pi.
Then I imported the tarball. The same result as of 3. above

leamas, 2023-05-28:
Still more work, this is not a simple oneliner to fix. But, I'm on it. please stay tuned.

Hakansv, 2023-08-02:
This particular issue is now solved.
Thanks

closed 2023-08-02 by Hakansv