OpenCPN/OpenCPN #3107: Route and Track API enhancements

antipole2, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
Managing routes by plugins is limited because few attributes of a route are accessible through the API.
I have previously raised this on FlySpray but Flyspray seems to have fallen out of use, whereas issues raised here get addressed.  Flyspray [#2842](https://opencpn.org/flyspray/index.php?do=details&task_id=2842&opened=1025&status%5B0%5D=) and [#2843](https://opencpn.org/flyspray/index.php?do=details&task_id=2843&opened=1025&status%5B0%5D=) refer.

**Describe the solution you'd like**
Please can we have the additional attributes in the route API, namely:
1. Priority
  *  .isVisible/Display on chart
  * Description
  * To / From
2. Desirable
  * Display parameters
  * Statistics (read only) - at least the distance

It would be good to have the same additions for tracks 

**Additional context**
The ability to read and set the visibility of waypoints is very useful.  I am working on the ability to manage the off-loading of waypoints by region/area. At present, it is not possible to do the same for routes or tracks.

bdbcat, 2023-03-26:
A bit late for this cycle, but since you are the only user of this API at present....
Routes:
I have added fields for reading and manipulating:
1. .isVisible/Display on chart
2. Description

To/From is already available.
I have not added line style, since this attribute is managed in some detail by the core classes.  Difficult at this point to coordinate the global interaction with this API.

Track:
No change.

We are very close (days) away from first Release Candidate.  Please test and report ASAP, please. 
Thanks

antipole2, 2023-03-26:
* isVisible/Display on chart
* Description

Both working.  Great!  Thanks a lot, Dave.

bdbcat, 2023-03-26:
We can close this issue now, and open a new one when we get past this Release of O58 ?

leamas, 2023-03-27:
@antipole2  ^

antipole2, 2023-03-27:
Yes close

closed 2023-03-27 by antipole2