OpenCPN/OpenCPN #3112: CPA toggle fine tuning

RooieDirk, 2023-03-26:
At start-up the canvasmenu shows always "CPA Alarm ->ON" , even if it is off.
Personally I would prefer to use the setting in the option menu to enable/disable the quick toggle setting. Maybe even hide/show. Where a switch in the Optionmenu from off -> on, always should set the toggle to ON and vv

Hakansv, 2023-03-26:
Dirk
The intention with the text  "CPA Alarm ->ON" is -If I click this item- the alarm will be set to "ON" --->On.
I'm sorry if it's misunderstood. In accordance with other canvas menu items the test would have been "Toggle CPA alarm" but then we have had to make a new Crowdin translation turn. To avoid that I used phrases already present in Crowdin. The logic I used was in beforehand tested by a couple of users, without explanation, and the response I got was positive.
Thoughts?

RooieDirk, 2023-03-26:
Morning Hakan,
How about using a checkbox??

Hakansv, 2023-03-26:
Hi.. 
Checkbox.. ? Hmmm
- No other canvas menu items use that for a toggle function.
- I'm not sure it's built for that menu?

RooieDirk, 2023-03-26:
[image]

Hakansv, 2023-03-26:
Ok - Nice check box. 
Still I think if introduced we have to for consistency reason add boxes also all other toggle functions for the different menu items.
But I agree if you found the present string confusing there would be others as well. (I may be "home blind" :) )
So what about this use of already present strings instead?
- "Show" "CPA alarm "
- "Hide" "CPA alarm "

RooieDirk, 2023-03-26:
That would be, for me as ignorant user, already much better

Hakansv, 2023-03-26:
I'd say the one who cares! 
Thanks, very good. I'll fix.
Good day --  Here's today summer time! Snow is falling. 2 C° below zero....

RooieDirk, 2023-03-26:
Only slightly warmer here, but at least a little sun.

closed 2023-03-26 by RooieDirk

bc38, 2023-03-26:
A controversial decision to stick everything in the Menu! The menu is already full.
Why don't you look at Chart Panel Options?, it even has an AIS section!

PR #3113 (2023-03-26): Update canvas menu CPA alarm text
Closes #3112