OpenCPN/OpenCPN #3114: Plugin local build with error can inhibit reinstallation of good plugin

antipole2, 2023-03-27:
**Describe the bug**
A plugin error seems to get noted that not cannot be cleared.

**To Reproduce**
Steps to reproduce the behaviour:

1. Do local build of plugin that will throw an error during init
2. Inport defective plugin
3. Enable plugin
4. Plugin appears to be enabled (ticked) but Preferences button is greyed out
5. Correct error in plugin, reimport and enable
6. Plugin still appears to be enabled (ticked) but Preferences button is greyed out and toolbar icon not inserted

*Expected behaviour**
It seems to remember it has had a duff plugin.  Reloading a correct plugin should clear the issue and load correctly.

**Desktop (please complete the following information if applicable):**
 - OS: macOS 
 - Version Catalina

**Additional context**
Have tried uninstalling plugin and also reinstalling from catalogue.  But once this error has occurred, unable to clear it.
However, restarting OCPN does clear it.

rgleason, 2023-03-27:
Yes, I have recently experienced this with javascript. I cleared all the plugins out from

- %localappdata%/opencpn/plugins 
- or C:\Users\fcgle\AppData\Local\opencpn\plugins 
- and from  C:\ProgramData\opencpn\plugins
- and from  C:\ProgramData\opencpn\plugins\cache\tarballs
- and from  C:\ProgramData\opencpn\plugins\cache\metadata

and was finally able to install javascript-2.1.3+186.290aad3_msvc-wx32-10-win32.tar.gz
successfully and show a JS icon rather than a green puzzle!
This had been plaguing me for quite awhile.  Rasbats could install the plugin successfully, but I could not.
It is now clear that I should have removed all previous instances of javascript from these locations.

rgleason, 2023-03-27:
Alec, for some reason, the failure of the previous plugin, prevents successful installation of the newer plugin, and requires the user to physically remove the older plugin (not sure which directory because I removed javascript from all of them).
I don't believe that is your intention. 

I understand your  "Part of this is https://github.com/OpenCPN/OpenCPN/issues/3103, a can of worms"  but we need to be aware of it.

antipole2, 2023-03-27:
A relief to know I am not losing it!  Just did not recognise the design when it caught me.
I am OK with this being closed if others are.

bdbcat, 2023-03-27:
Using the normal catalog installation workflow, "real" users will not likely see plugins that fail during initialization stage.  Also, IMHO, average users will not use the plugin import feature often, if at all.
So I think this behavior is satisfactory for now, may be improved in future releases.
Close it.

leamas, 2024-02-07:
The ideal solution here would be that importing a new plugin version after a failed attempt would just work. However, here is a chicken and egg problem. The information telling a plugin is imported is in the metadata, but to find the metadata for a plugin one needs to load it first. Since the plugin in this case is blacklisted, this fails.

The blacklisting is for good reasons: plugins are reloaded in a lot of cases, and we only want the message about a plugin being unloadable once per session. 

This leaves us with two options:
1. Leave things as they are and close bug, basically as WONTFIX
2. Clear all blacklist entries when importing a plugin.

2. means that if there are "bad", unloadable  plugins installed the message for them will be repeated after an import operation.  This is not convenient. OTOH, it solves the problem in this issue and makes it possible to import the next plugin build. Perhaps it's worth it.  I'll make a PR along these lines.

leamas, 2024-02-07:
Now, if anyone feels motivated I need help testing this. If there is such input we should handle it.

If there is no such input my plan is to close both this PR and #3114 without further actions.

leamas, 2024-02-07:
@antipole2 ^

antipole2, 2024-02-08:
Please can someone remind me what kind of plugin error was causing this issue.
If it is a missing symbol and the plugin gets quarantined, I only need to fix the missing symbol and re-import and problem goes away.
I recall the issue was not a missing symbol but some kind of error during init which was not sufficient to crash OCPN.

Am using OCPN 5.8. Let me know if I need recent build for test.

leamas, 2024-02-08:
You need to build the PR to test, it is not merged.

No, I don't remember what kind of plugin problem you had ;) 

But the new logic should step in if you first make a plugin build with for example a missing symbol. In master, this means that if if you rebuild the plugin without the missing symbol it is not imported unless you restart opencpn. Using the PR, the new build should be possible to import without a restart.

antipole2, 2024-02-09:
I am not set up to build OCPN - never needed to.  Any way around this?
Yes, I could follow the recipes and get set up, but I am short of time (in cruise planning mode).

nohal, 2024-02-09:
No way a round it, pull request builds are not published.
If you are locally building plugins, you should be set up for building OpenCPN.

The easiest way to do local build of OpenCPN is to use dependencies from Homebrew

```sh
brew install cmake
brew install gettext
brew install lame
brew install lz4
brew install mpg123
brew install xz
brew install zstd
brew install libarchive
brew install wxwidgets
```

To generate the Xcode project

```sh
cd /opencpn/source
mkdir build
cd build
cmake -G Xcode  ..
```

Then either open it from Xcode manually or still in the same terminal run `open OpenCPN.xcodeproj` and it will be started for you. In the Xcode GUI select the `OpenCPN` target from the dropdown in the middle top of the screen and hit the "play button", the application gets built and started with a debugger attached to it.

antipole2, 2024-02-17:
@nohal I have at last got around to trying this - first without the commit, so building with master.
Your recipe above worked well, except that on linking I get missing symbols and
> Search path '/usr/local/lib/Debug' not found

Maybe I need to run a further brew install ?

nohal, 2024-02-17:
I really don't see what it should be looking for in `/usr/local/lib/Debug`. Can I see more of the log?

antipole2, 2024-02-17:
[Build tests_2024-02-17T16-33-16.txt.zip](https://github.com/OpenCPN/OpenCPN/files/14319651/Build.tests_2024-02-17T16-33-16.txt.zip)

nohal, 2024-02-17:
Your problem is `libarchive` :
```
ld: warning: ignoring file '/usr/local/Cellar/libarchive/3.4.2_1/lib/libarchive.13.dylib': found architecture 'x86_64', required architecture 'arm64'
ld: Undefined symbols:
  _archive_entry_filetype, referenced from:
...
```

Your system is ARM, so I wonder what does some relic of Homebrew for x86_64 do there...

antipole2, 2024-02-17:
Here is the libarchive install log and it does mention multiple versions.  Shall I delete the existing libarchives and brew again?
```
brew install libarchive
==> Downloading https://ghcr.io/v2/homebrew/core/libarchive/manifests/3.7.2
############################################################################################## 100.0%
==> Fetching dependencies for libarchive: libb2
==> Downloading https://ghcr.io/v2/homebrew/core/libb2/manifests/0.98.1
############################################################################################## 100.0%
==> Fetching libb2
==> Downloading https://ghcr.io/v2/homebrew/core/libb2/blobs/sha256:52cef2730b3520e99f75f1478f2b953dc
############################################################################################## 100.0%
==> Fetching libarchive
==> Downloading https://ghcr.io/v2/homebrew/core/libarchive/blobs/sha256:5dcd7bf0a4b9ed437b6e0c6b19f3
############################################################################################## 100.0%
==> Installing dependencies for libarchive: libb2
==> Installing libarchive dependency: libb2
==> Downloading https://ghcr.io/v2/homebrew/core/libb2/manifests/0.98.1
Already downloaded: /Users/Tony/Library/Caches/Homebrew/downloads/9bb23dca331f0d74876fa5c769113885b92af8a741a851cdfe99f6dfd420b12d--libb2-0.98.1.bottle_manifest.json
==> Pouring libb2--0.98.1.arm64_sonoma.bottle.tar.gz
🍺  /opt/homebrew/Cellar/libb2/0.98.1: 8 files, 126.4KB
==> Installing libarchive
==> Pouring libarchive--3.7.2.arm64_sonoma.bottle.tar.gz
==> Caveats
libarchive is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

If you need to have libarchive first in your PATH, run:
  echo 'export PATH="/opt/homebrew/opt/libarchive/bin:$PATH"' >> ~/.zshrc

For compilers to find libarchive you may need to set:
  export LDFLAGS="-L/opt/homebrew/opt/libarchive/lib"
  export CPPFLAGS="-I/opt/homebrew/opt/libarchive/include"

For pkg-config to find libarchive you may need to set:
  export PKG_CONFIG_PATH="/opt/homebrew/opt/libarchive/lib/pkgconfig"
==> Summary
🍺  /opt/homebrew/Cellar/libarchive/3.7.2: 64 files, 4.0MB
==> Running `brew cleanup libarchive`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
==> Caveats
==> libarchive
libarchive is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

If you need to have libarchive first in your PATH, run:
  echo 'export PATH="/opt/homebrew/opt/libarchive/bin:$PATH"' >> ~/.zshrc

For compilers to find libarchive you may need to set:
  export LDFLAGS="-L/opt/homebrew/opt/libarchive/lib"
  export CPPFLAGS="-I/opt/homebrew/opt/libarchive/include"

For pkg-config to find libarchive you may need to set:
  export PKG_CONFIG_PATH="/opt/homebrew/opt/libarchive/lib/pkgconfig"
```

nohal, 2024-02-17:
This one is not the problem, the one you for some reason have in `/usr/local/Cellar/` is. This should not exist at all on your system.

antipole2, 2024-02-17:
I tried brewing libarchive again and got
```
libarchive is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.
```
So I removed the one in /usr/local/Cellar/
I then had to use reinstall and I got
```
libarchive is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

If you need to have libarchive first in your PATH, run:
  echo 'export PATH="/opt/homebrew/opt/libarchive/bin:$PATH"' >> ~/.zshrc
```
So I did the last PATH update and started back at `cmake -G Xcode  ..`
Now the xCode build fails with 
```
Showing All Issues
No such file or directory: '/usr/local/opt/libarchive/lib/libarchive.dylib'
```
Clearly still missing something in the right place.

nohal, 2024-02-17:
Did you clean the build directory before that?

antipole2, 2024-02-17:
Ah!  I had seen OpenCPN to the left but that was name of folder not the scheme.
I have now launched OCPN and it has immediately got an address error in loadPluginCandidate.
I guess that is because I am yet to try the branch with the patch for this.
How far back do I need to go when a source is changed?  Can I just relaunch in Xcode?

nohal, 2024-02-17:
Sorry, I don't understand the question. What branch a re we talking about? Product of this build normally hosts any plugins from  the master catalog. I f you have some plugin dylib somewhere that makes it crash, we will probably want to see it.
If the source of OpenCPN changes, you just `git pull`, rebuild and are done.

antipole2, 2024-02-17:
I am doing this to test a PR [see above](https://github.com/OpenCPN/OpenCPN/issues/3114#issuecomment-1933813685).
You asked me to test because I have had a bad plugin such that  my good plugin will not load into v5.9 but does in 5.8.
I have started by building from Master so I get the process right.

Now I need to do a PR to test the mod.  Not sure how I do that. Is it on a separate branch or what?
`https://github.com/OpenCPN/OpenCPN/pull/3635`

leamas, 2024-02-18:
> Is it on a separate branch or what?

It is is another repositry. Without explanations, the step to check it out is:

    $ git remote add leamas  https://github.com/leamas/OpenCPN.git
    $ git remote update leamas
    $ git fetch leamas reload-3114:reload-3114
    $ git checkout reload-3114

After these steps you have checked out the reload-3114 branch from my repo.

---
Notes:

 If you look into https://github.com/OpenCPN/OpenCPN/pull/3635 you find the message _[leamas](https://github.com/leamas) wants to merge 2 commits into [OpenCPN:master](https://github.com/OpenCPN/OpenCPN/tree/master) from [leamas:reload-3114](https://github.com/leamas/OpenCPN/tree/reload-3114)_. This is where I got the information about which branch to use.

If you follow the   [leamas:reload-3114](https://github.com/leamas/OpenCPN/tree/reload-3114) link you find the green Code button. This is where I found the address for the remote.

leamas, 2024-04-28:
Seeing #3832 being resolved I think we should close this bug, basically as WONTFIX. If anyone disagree, this is the time to speak up.

bdbcat, 2024-04-28:
Agree  WONTFIX

antipole2, 2024-04-28:
Maybe this has resolved with the various changes.  Let’s close and worry about it if it shows up again.

closed 2024-04-28 by antipole2