OpenCPN/OpenCPN #3117: Canvas scaling wrong on macOS Retina display in non-OpenGL mode

nohal, 2023-03-29:
If OpenGL is turned of on macOS system with Retina display
- Chart canvas scale is wrong
- Chart bar and scale indicator missing
- Canvas size is reported too big to plugins

Seems like the calculations are done using the retina pixels, but the display is actually using DIP in non-GL mode.

non-GL, broken
[image]

OpenGL, correct (only action between the two screenshots is switching OpenGL on)
[image]

bdbcat, 2023-03-30:
This is turning out to be a can of worms....
Turns out the OpenGL scaling is wrong.  Prove by drawing a 100 NM route, and measure the size on screen (with a ruler).  Do the math.  Wrong by 2x. on GL mode.
Changes to fix will be pervasive, including o-charts_pi....sigh...
Makes me nervous at Release Candidate time, even though should be confined to MacOS.
I'm on it, don't know where it will lead.

bdbcat, 2023-03-31:
Pavel...
I merged a change to somewhat correct the non-GL rendering on Mac Retina display.
TBH, the code is not pretty, and performance in this mode is poor.  Further, there is an unintentional scale change when shifting display modes, and other problems with chart rendering.
Further improvement will need to wait for a later releases.  There is much re-factoring to do...but not now.

bdbcat, 2023-03-31:
So, can we remove this as a blocker for O58,  and mark for future O5.10?  It really will take some serious refactoring to get it right without a bunch of #ifdef __WXOSX__.  Who likes that?

nohal, 2023-03-31:
The plugin part sure. What about the 2x scale problem in GL mode though (I suppose most users on macOS use GL)? Make it another issue?

bdbcat, 2023-03-31:
The _reported_ scale in GL mode is just wrong by 2X.  This I don't mind fixing this reported value by #ifdef hack, since it is only informational on screen.
But the actual GL display is nice, and the underlying GL code understands the display extents correctly in GL mode.
The non-GL case is more trouble, as you noted.
So I suggest we "fix" the GL case, and pass on the non-GL case for now.  I cannot think of a use-case where non-GL would be better on MacOS.

nohal, 2023-03-31:
Agreed, the non-GL mode on macOS sure can wait for next release.