OpenCPN/OpenCPN #3121: Route | Send to peer: Question: "Ok to overwrite" in wrong place

leamas, 2023-03-31:
**Describe the bug**
When trying to transfer a route to a peer and the route already exists, the process seems to just hang. The reason is that the remote peer has a dialog "OK to overwrite" and waits until it's completed. This could be quite confusing from the perspective of the user issuing "Send to Peer".

**To Reproduce**
Steps to reproduce the behavior:
1. Start OpenCPN, make sure there is at least one route available
2. Start another OpenCPN instance on the same network
3. On the original box, right-click on a route and select "Send to Peer". Answer questions about peer to use and also pin code. This works as expected
4. Repeat: right click on route, do "Send to Peer" using the same peer.
7. Nothing happens (well, it does happen, but somewhere else).

**Expected behavior**
The  "OK to overwrite" dialog should occur on the machine which initiated the transfer. This is based on that our basic use case is that both machines are used by the same user.

**Additional context**
Way to late for 5.8. Perhaps we could add a simple "always overwrite" configuration option without UI for the time being.

leamas, 2023-03-31:
I missed the "Always replace objects from this source" which sort works around this issue for now. 

Even so, the bug is still here but with less impact.

Hakansv, 2023-03-31:
I can agree on task 5 above but wouldn't call it a bug. It's working as expected. Maybe a future request where I can agree.
While the client is waiting for a server response a status message like: "Awaiting peer acceptance" would make it clear. That implies another turn of client-server messaging. 
I though don't think it so important we need to brake the RC1 process with new translations and more. We better wait?

leamas, 2023-03-31:
>  It's working as expected

I disagree. If you are initiating a workflow on a machine things are not supposed to just stall on that. More generally,  you need some very good reasons to split the workflow  "upload route" to two different machines, reasons I cannot see. Can you?

> While the client is waiting for a server response a status message like: "Awaiting peer acceptance" would make it clear. That implies another turn of client-server messaging.

Any solution  to this indeed requires  a new  message passing cycle with the peer.  Since this is needed anyway, why not do the natural thing to raise the question on the machine causing it?

> I though don't think it so important we need to brake the RC1 

Already stated in the initial description, see _Additional Context_

leamas, 2023-03-31:
> We need to rethink this

My initial idea is that before actually transferring the route the initiating peer sends a question like "OK to update/overwrite route foo?"  to the remote peer. Depending on the reply it then  either displays the "OK to overwrite" dialog or just sends the route.  

The remote peer then just trusts incoming route updates to be ok.

leamas, 2023-12-30:
Server part of this is in place after #3520.

PR #3584 (2024-01-16): Peer transfers: client overhaul.
Client side of the peer transfer code.
  - Refactor peer_client into somewhat proper model + gui parts.
  - Fix the "Ok to overwrite dialog" so it occurs on the originating peer rather than the server. This requires that both client and server is updated.
  - Fix bug leading to duplicated waypoints on the receiving side. This is a server only fix not requiring an updated client.

All sorts of code clean up and GUI polish. Some memory leaks was detected and plugged.

Closes: #3121
Closes: #3522 

EDIT: Added new commits handling activation of a transferred route [... cut]
closed 2024-01-16 by bdbcat