OpenCPN/OpenCPN #3126: Properties window disappears behind canvas on MacOS (RC1)

antipole2, 2023-04-03:
**Describe the bug**
When viewing a waypoint, route or track properties window, it disappears behind the frame whenever the frame gains focus

**To Reproduce**
On MacOS
1. Open a waypoint properties window and position it over the canvas
2. Douchte canvasor hover over the Toolbar
3. The properties window disappears behind the canvas.

**Expected behavior**
It should stay on top.

**Desktop (please complete the following information if applicable):**
 - OS:  macOS
 - Version Ventura

**Additional context**
Need to turn on the wxSTAY_ON_TOP property when MacOS

leamas, 2023-04-03:
hm... Trying the latest build 5.7.1+6ffefdb on a fresh ventura install..  Here it seems that the properties window stays on top even if the canvas get's focus. 

And: I actually find this behaviour very disturbing. My personal expectation is that any non-modal window which gets focus should be on top. That is, I **want** the canvas to cover the properties window if  I click in it to give it focus.

So, many questions here:
- Has anything changed w r t this  lately? Or is this is a difference between my and your Ventura installation?
- Are there any Apple UI guidelines supporting your claim that the expected behaviour is that the properties windows should stay on top?

leamas, 2023-04-04:
1

leamas, 2023-04-04:
oops... I didn't mean to close it, nor did I mean to make that last comment.

 Sorry. Reopening

antipole2, 2023-04-04:
The latest build of the JavaScript plugin (in my alpha repository) has an option in the Consoles tab of the plugin Tools to turn on or off wxSTAY_ON_TOP for all consoles.  So you can experiment with the effect on each platform.
For Mac, I prefer it on but you can turn it off if that's how you want to work.

nohal, 2023-04-04:
We have been there, at the end `wxSTAY_ON_TOP` makes more harm than good AFAICT. The "correct" behavior on macOS simply is that non-modal windows get bellow when they don't have focus. And the only reasonable workaround is to use `wxFrame` floating on parent.

antipole2, 2024-04-06:
For the record, JavaScript_pi v2.2 switched from using wxDialog to wxFrame for its consoles.  This was easier than I anticipated.  It was a huge improvement. Together with the user preference to turn on or off wxSTAY_ON_TOP according to taste, I no longer have an issue here.

antipole2, 2024-07-02:
Closing this as no longer an issue and accept nohal's position,

closed 2024-07-02 by antipole2