OpenCPN/OpenCPN #3128: Window asking whether to disable plugins needs to stay on top (MacOS)

antipole2, 2023-04-05:
**Describe the bug**
When OCPN is restarted after an untidy end, it displays a window asking if plugins should be disabled.
Sometimes, under MacOS, this window opens behind the canvas and is not apparent.
OCPN appears to be hanging with the watch cursor indicating it is busy but no indication the user must look behind the canvas.

**To Reproduce**
Steps to reproduce the behavior:
1. Crash OCPN
2. Restart

**Expected behaviour**
Question window should always be visiible.

**Desktop (please complete the following information if applicable):**
 - OS: Likely a MacOSonly issue

**Additional context**
Having has some crashes (reasons not relevant here) I feared I had done some serious mischief as OCPN appeared to hang on restart before I had a chance to disable possibly misbehaving plugins.  As I had been performing big changes on waypoints and routes, I feared I had screwed up something seriously.  Eventually, I discovered the restart question behind the canvas.
Ordinary users could easily be caught by this.

transmitterdan, 2023-04-05:
While I am sure this is frustrating, in the end this dialog pop-up will self-extinguish after a time.  So patience will be rewarded in the end.

leamas, 2025-05-03:
I have tested recent builds on MacOS. The dialog is indeed modal, the chart canvas si not displayed until the dialog is handled.

Will close this issue unless there is more input in a day or two.

antipole2, 2025-05-03:
Agree - it is now displayed before the canvas.  All looks good.

closed 2025-05-03 by leamas