OpenCPN/OpenCPN #3129: Do not erase user's tidal file on a O upgrade

Hakansv, 2023-04-06:
**Describe the bug**
When OCPN is upgraded to a new version all tidal files are erased.
OCPNPlatform.cpp:1369
```
// Force a reload of updated default tide/current datasets
    TideCurrentDataSet.clear();
```
I suppose the intention is to clear out outdated Harmonics from earlier O.
But by "clear() all files are deleted also those a user may have added to cover more than what's included default.
**To Reproduce**
Steps to reproduce the behavior:
1. Update OCPN to a new version

**Expected behavior**
On a O upgrade clear only files installed from O default "tcdata". 
Leave what's may be added from other file locations.

**Screenshots**
[See this CF post](https://www.cruisersforum.com/forums/f134/opencpn-version-5-7-1-release-candidate-2-rc2-274552.html#post3763665)

bdbcat, 2023-04-06:
I propose to defer this until next version.
1.  The tide dataset file management needs work.  Any quick fix would be risky.
2. Affects a relatively few number of real users.

Hakansv, 2023-04-06:
Agreed. Both on deferment and that a fix of this needs more test than we have time for.

leamas, 2023-04-06:
Just a note for the future: this might be an pilot for the kind of data-only plugins we have tools in place for....

Hakansv, 2023-05-25:
Corrected by 6d15424f7ee

closed 2023-05-25 by Hakansv