OpenCPN/OpenCPN #3130: Plugins on Ubuntu 22.04 (5.7.1) using wrong flavor(?)

LennartG-Sve, 2023-04-06:
Installing plugins on Ubuntu 22.04 (5.7.1) fails with error message: Plugin depends on library "libwx-baseu-3.0.so.0", but the main module was built for "libwx-baseu-3.2.so.0".

Using CatalogExpert and changing the flavor to 'ubuntu-wx32-x86:22.04', that appears to be the closest relative, allows almost all plugins loaded from both master and beta to load correctly. There is one (two?) exception(s), ShipDriver failed to load, both the master and beta versions. No error message regarding libwx was given in the OpenCPN.log.

Interesting as 'ubuntu-wx32-x86:22.04', as I understand it, should be the default flavour.

leamas, 2023-04-06:
hm... This might be a build error, a missing option to cmake when building the package. 

*'ubuntu-wx32-x86:22.04* is indeed the indended target. However, one question: Are you sure that you have not been using the CatalogExpert earlier to set things? That is, can you reproduce these results after removing the lines containing `CompatOS=` and `CompatOsVersion=`  from ~/.opencpn/opencpn.conf?

LennartG-Sve, 2023-04-06:
@leamas: I had never used the flavor setting before even though CataloExpert was used to select the beta catalog. You may find more info under testpkugin issue #299.
What flavor setting should I use for the --loglevel debug option (if still required)?

leamas, 2023-04-06:
Could you please reset it to *Default setting*, one of the first options in the combobox? And yes, getting your opencpn.log  with *--loglevel debug* would be very useful.

LennartG-Sve, 2023-04-06:
Not really surprised, could well be something that Dave has fixed in the meantime, My 5.7.1 is about three weeks old. 
Also noted that the default setting now does not conform to the default I had before I started these tests (ubuntu-x86_64:2204) but the wx32 version I later switched to.
The ubuntu-wx-32-x86_64:2204 is also what I get now when selecting 'Default'
Repeated some load tests on e.g. Logbook and Watchdog that failed before, they both worked now. Shipdriver however fails but Rasbats is on that one.
Case solved (?).

leamas, 2023-04-06:
Yes, solved. Closing, feel free to insult me if I'm wrong.

closed 2023-04-06 by leamas