OpenCPN/OpenCPN #3132: Enable ECMWF in grib plugin ?

trudK45, 2023-04-08:
Are there any plans adding the download option for the European ECMWF model ?
Available parameters are : WIND, MSLP, HGT500, TEMP (2m air temp) and WAVES (sig wave height)
I have tried these by patching the email sent to saildocs and they work.

https://saildocs.com/gribmodels

nohal, 2023-04-08:
Isn't this already implemented since https://github.com/OpenCPN/OpenCPN/pull/3045 ?

trudK45, 2023-04-08:
So this was added just the other day.
I will have to download and build I suppose.

trudK45, 2023-04-09:
Which version of wxWidgets should I use, wxWidgets.3.2.2 ?

nohal, 2023-04-09:
Yes, OpenCPN 5.8.0 uses wxWidgets 3.2.2 in all official packages.

trudK45, 2023-04-09:
I am using Visual Studio 2017. Do I need to upgrade to 2022 ?

Runing config I get these errors.

-- Could NOT find wxWidgets (missing: gl net xml html adv aui core base webview)
-- Could not find wxWidgets webview add-on
-- Could NOT find wxWidgets (missing: gl net xml html adv aui core base)
CMake Error at CMakeLists.txt:755 (message):
  Cannot find wxWidgets, giving up.  On linux you might want to try using
  -DOCPN_FORCE_GTK3.

nohal, 2023-04-10:
You should certainly follow the instructions from https://opencpn-manuals.github.io/main/opencpn-dev/windows.html#_5_8_0_fast_track or simply download the CI packages from https://cloudsmith.io/~david-register/repos/opencpn-unstable/packages/

trudK45, 2023-04-10:
Thanks Pavel!

I downloaded opencpn_5.7.1+386.325505d

For ECMWF the available parameters at this time are WIND, MSLP, HGT500, TEMP (2m air temp) and WAVES

From what I can see WAVES for ECMWF is not included as selection.

WIND500 is valid for GFS but is it really that for ECMWF ?
It might be, as I get no errors from Saildocs when I send it include it in the request.

/Hans

bc38, 2023-04-13:
[trudK45](https://github.com/trudK45)
This is a limited free version of the ECMWF model

trudK45, 2023-04-13:
> [trudK45](https://github.com/trudK45) This is a limited free version of the ECMWF model

Yes, and WAVES are included in the limited free version of the ECMWF model.

This is what I get swnding a request for ECMWF when I patch the request with WAVES.

Grib extracted from file ecmwf-20230409-00z.grb dated 2023/04/09 07:55:30
request code: ECMWF:39N,23N,28W,6W|0.25,0.25|0,3,6..192|WIND,PRESS,MSLP,WAVES

NOTICE:
From 8-Mar 2022, ECMWF "Essential" is replaced with ECMWF "Open Data", now available without charge. Parameters are limited compared to full ECMWF access and the data is delayed about one hour.  

Global forecast data is available from 0-144 hours at 3-hour increments, and 150-240 hours at 6-hour increments. The data grid is 0.4-deg, and available parameters at this time are WIND, MSLP, HGT500, TEMP (2m air temp) and WAVES (sig wave height). Data is updated every 12 hours, and available by 8 hours after the synoptic time e.g. 08:00 utc for the 00z model-run. The request code remains "ECMWF".

LeaseOnLife, 2023-04-19:
In other words: the ECMWF request by the grib plugin currently does not allow selecting "WAVES". the other available parameters do work. Only "WAVES" needs to be added.
[image]

trudK45, 2023-04-19:
> In other words: the ECMWF request by the grib plugin currently does not allow selecting "WAVES". the other available parameters do work. Only "WAVES" needs to be added.

Correct.

closed 2023-04-21 by bdbcat

bdbcat, 2023-04-25:
@nohal 
Did you mean for this, and associated color palette update for dark mode, to be in v5.8.x?

nohal, 2023-04-25:
Yes

bdbcat, 2023-04-25:
OK, thanks.  Getting close to the bottom of the 582 list.

PR #3156 (2023-04-21): Grib plugin fixes
- Add WAVES option to the ECMWF request. Fixes #3132
- Fix the meteotable colors on systems with dark theme
Old colors:
[image]
New colors:
[image]
commit 9eedb806 (2023-04-21): Add the WAVES parameter to ECMWF grib request to Saildocs. Fixes #3132
commit 20287861 (2023-04-25): Add the WAVES parameter to ECMWF grib request to Saildocs. Fixes #3132