OpenCPN/OpenCPN #3137: RFE: Add support for Victron MQTT

mgrouch, 2023-04-14:
Along with dashboards
For power, tanks, temperatures etc. 

Thanks

mgrouch, 2023-11-15:
Some code from 

https://github.com/bareboat-necessities/bbn-m5stack-tough

could be reused for that. 

Specifically 
https://github.com/bareboat-necessities/bbn-m5stack-tough/blob/main/bbn_m5tough_active_boat/net_mqtt.h

And possibly LVGL for UI gauges and widgets.

mgrouch, 2024-01-26:
Any chance someone in OpenCPN dev looked at it? It would be very helpful addition 

thanks

nohal, 2024-01-26:
This is out of scope for OpenCPN as a chartplotter. Dashboard integrated in OpenCPN is already possible via Signal K with the DashboardSK plugin. To actively interact with Victron equipment a separate plugin serving that purpose is the way to do it.

aviatorhh, 2024-04-29:
I myself am using Victron equipment in that way described by Pavel and it works just fine (just reading, no switching).
I agree that this is a plugin feature (request). In my opinion it would only avoid a Signal K server and use a Certbot or other MQTT brokers directly. I have tried to avoid Signal K in the past myself not to have too much stuff running but in the end it helped even to simplify other things. 
So from my perspective this issue can be closed?
DashboardSK is a good basis for displaying additional data in O. I think it would be more efficient to develop other gauge types/styles in future.