OpenCPN/OpenCPN #3144: Visual indication of chart group

nohal, 2023-04-18:
There are often users who switch to a chart group without realizing that and consider the rest of the charts as not working or lost.

The last published occurrence: https://www.cruisersforum.com/forums/f134/mbtiles-not-displaying-274940-2.html#post3767790

**Describe the solution you'd like**

We should visually indicate that a chart group is active.

bc38, 2023-04-18:
There is also a label in the form of a dot in the group selection menu. Maybe make it bigger?

nohal, 2023-04-18:
Even if possible, that won't help much as the users do not usually look there as we can see from the reports.

bc38, 2023-04-18:
I noticed questions from the author on the forum, they are all quite simple. The answers to which are in the manual. He didn't even have a group.
It may be worth adding to the manual so that displaying mbtiles maps of a certain category, for example, satellite images in its pure form, is possible only if the mbtiles files are collected in one folder, added to a directory and assigned a group for this directory called Satellite images.

nohal, 2023-04-18:
This has nothing to do with MBTiles, this is a general problem that we need to address. I was helping several boats in Antarctica this season who "lost" their S63 charts for exactly the same reason.

bdbcat, 2023-04-18:
What sort on GUI indication comes to mind?  Something in the status bar?  Or some persistent item on screen like e.g. the scale bar?

nohal, 2023-04-18:
I was thinking about maybe adding some icon at the beginning of the piano bar.

andyrbarrow, 2023-04-18:
Can I suggest just putting the name of the chart group on the bar? If there is a concern about too many characters, just trim the name and put "..." at the end. As an active user, and someone who supports other new users locally, I'm thinking that even putting a file folder icon on the bottom won't be enough of a message for them.

Hakansv, 2023-04-21:
> Can I suggest just putting the name of the chart group on the bar?

I can personally agree on that. But maybe not the name since the bar can sometimes be rather crowded. 
Instead, if not "All"; the chart group number, [CG1] / [CG2]. And the group name as the tool tip?

[image]

rgleason, 2023-07-31:
@bc38 have added this to chart groups in the manual
"It may be worth adding to the manual so that displaying mbtiles maps of a certain category, for example, satellite images in its pure form, is possible only if the mbtiles files are collected in one folder, added to a directory and assigned a group for this directory called Satellite images."

Thanks