OpenCPN/OpenCPN #3146: toolbar flickering when expanded

bensasl2, 2023-04-18:
**Describe the bug**
the toolbar is flickering when expanded using the hamburger icon. 
**To Reproduce**
shrink the toolbar then expand it using the hamburger icon.

**Expected behavior**
effectively minimize or resolve flickering issues in the wxWidgets toolbar

**Desktop (please complete the following information if applicable):**
 - OS: Windows 
 - Version 11

**Additional context**
flickering can be a common issue in user interface elements developed using the wxWidgets framework. However, wxWidgets provides several mechanisms to help mitigate or eliminate flickering, such as double buffering, optimized drawing functions, and platform-specific optimizations.

bensasl2, 2023-04-19:
[Recording 2023-04-19 071545.zip](https://github.com/OpenCPN/OpenCPN/files/11269329/Recording.2023-04-19.071545.zip)

leamas, 2023-04-19:
@bensasl2 : You are planning to work with this issue, right?

bensasl2, 2023-04-19:
yes I am already working on this UI improvement. I thought It's worth checking if anyone is aware and had an idea or half a solution...

leamas, 2023-04-19:
It certainly is. Assigning issue to you to make it all clear.

bdbcat, 2023-04-19:
There is code in the toolbar management to tries to smoothly "roll up" the toolbar when hidden.  Works OK on reasonably fast hosts.  Not as well on slower devices.
Have a look around ocpn_frame:2649

bensasl2, 2023-04-19:
When I simply comment the following line : ocpn_frame.cpp:2816
the toggle behaviour becomes much better in my opinion ( see recording attached) 
[Recording 2023-04-19 225533.zip](https://github.com/OpenCPN/OpenCPN/files/11277879/Recording.2023-04-19.225533.zip)

I suspect this line was added by mistake anyway because a comment just below in the code says :
      //  One last "Realize()" to establish the final toolbar shape
so the Realize() function only meant to be called once at the end of the animation instead of each time the tool count is changed otherwise it causes flickering.

bdbcat, 2023-04-20:
Well, maybe...
As I recall, this function had different problems on different platforms.  Will need to be carefully tested on Windows/Mac/Rpi/linux before general adoption.

leamas, 2024-04-01:
@bensasl2 : Major work with the toolbars as been done #3593. As I understand it, this should resolve also this issue. Can you confirm this?

leamas, 2024-04-03:
Given recent work in #3579 and #3593 I assume that this issue also is fixed. If not, please add a note here and I'll reopen.

Closing

closed 2024-04-03 by leamas