OpenCPN/OpenCPN #3150: Crash at shutdown after changing language

nohal, 2023-04-19:
**Describe the bug**

If language is changed during the session, application crashes on shutdown.

**To Reproduce**

- Have some plugins
- Go to Toolbox -> User Interface
- Change language
- Exit the application
- See the crash
```
#0  0x00007ffff6f24300 in wxRefCounter::DecRef() ()
    at /lib64/libwx_baseu-3.2.so.0
#1  0x00007ffff6f2439e in wxObject::UnRef() () at /lib64/libwx_baseu-3.2.so.0
#2  0x00007ffff6eac043 in wxObject::~wxObject() ()
    at /lib64/libwx_baseu-3.2.so.0
#3  0x0000000000a87676 in PlugInContainer::~PlugInContainer()
    (this=0x5d63530, __in_chrg=<optimized out>)
    at /home/nohal/source/OpenCPN/include/plugin_loader.h:84
#4  0x0000000000a7dbcd in PluginLoader::UnLoadPlugIn(unsigned long)
    (this=this@entry=0x1377c70, ix=ix@entry=0)
    at /home/nohal/source/OpenCPN/src/plugin_loader.cpp:632
#5  0x0000000000a7dcea in PluginLoader::UnLoadAllPlugIns() (this=0x1377c70)
    at /home/nohal/source/OpenCPN/src/plugin_loader.cpp:818
#6  0x00000000007cf46d in MyFrame::OnCloseWindow(wxCloseEvent&)
    (this=0x1379640, event=<optimized out>)
    at /home/nohal/source/OpenCPN/src/ocpn_frame.cpp:1876
#7  0x00007ffff6fc41b2 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) () at /lib64/libwx_baseu-3.2.so.0
#8  0x00007ffff6fc58d3 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /lib64/libwx_baseu-3.2.so.0
#9  0x00007ffff6fc6a76 in wxEvtHandler::TryHereOnly(wxEvent&) ()
    at /lib64/libwx_baseu-3.2.so.0
#10 0x00007ffff6fc437e in wxEvtHandler::DoTryChain(wxEvent&) ()
    at /lib64/libwx_baseu-3.2.so.0
#11 0x00007ffff6fc6c12 in wxEvtHandler::ProcessEvent(wxEvent&) () at /lib64/libwx_baseu-3.2.so.0
#12 0x00007ffff6fc77c3 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /lib64/libwx_baseu-3.2.so.0
#13 0x00007ffff776e8fd in wxWindowBase::Close(bool) () at /lib64/libwx_gtk3u_core-3.2.so.0
#14 0x00007ffff75a62aa in gtk_frame_delete_callback () at /lib64/libwx_gtk3u_core-3.2.so.0
#15 0x00007ffff609cc27 in _gtk_marshal_BOOLEAN__BOXEDv () at /lib64/libgtk-3.so.0
#16 0x00007ffff5b66608 in g_signal_emit_valist () at /lib64/libgobject-2.0.so.0
#17 0x00007ffff5b666f3 in g_signal_emit () at /lib64/libgobject-2.0.so.0
#18 0x00007ffff636f9f4 in gtk_widget_event_internal.part.0.lto_priv () at /lib64/libgtk-3.so.0
#19 0x00007ffff6208ee5 in gtk_main_do_event () at /lib64/libgtk-3.so.0
#20 0x00007ffff6ca5677 in _gdk_event_emit () at /lib64/libgdk-3.so.0
#21 0x00007ffff6cf74ee in gdk_event_source_dispatch.lto_priv () at /lib64/libgdk-3.so.0
#22 0x00007ffff5a45f58 in g_main_context_dispatch () at /lib64/libglib-2.0.so.0
#23 0x00007ffff5aa5cd8 in g_main_context_iterate.isra () at /lib64/libglib-2.0.so.0
#24 0x00007ffff5a4bbcf in g_main_loop_run () at /lib64/libglib-2.0.so.0
#25 0x00007ffff6206305 in gtk_main () at /lib64/libgtk-3.so.0
#26 0x00007ffff758a455 in wxGUIEventLoop::DoRun() () at /lib64/libwx_gtk3u_core-3.2.so.0
#27 0x00007ffff6ed6c1a in wxEventLoopBase::Run() () at /lib64/libwx_baseu-3.2.so.0
#28 0x00007ffff6eb300d in wxAppConsoleBase::MainLoop() () at /lib64/libwx_baseu-3.2.so.0
#29 0x00007ffff6f13cd0 in wxEntry(int&, wchar_t**) () at /lib64/libwx_baseu-3.2.so.0
#30 0x00000000005dc0b8 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at /home/nohal/source/OpenCPN/src/ocpn_app.cpp:935
```

transmitterdan, 2023-04-20:
@nohal I have confirmed this is repeatable and I believe it is related to #3151 

On Windows I can get a crash every time on exit if I change the locale language and then close O.  The indication is heap corruption.  I cannot see what is trashing the heap.  The crash happens in the plugin manager as it is deleting 'pic' items.  Specifically it is deleting a wxBitmap that ultimately causes the crash but I think that is because the wxBitmap pointer has been trashed.

nohal, 2023-04-20:
#3151 is confined to a single account on a single system. Not reproducible under different user even on that very machine. And happens already when the form is being created, you never even get to changing the locale.
I don't think they are related in any way but having something to do with that can of worm the way we handle the translations is.

leamas, 2023-04-20:
> wxWidgets does not use smart pointers. 

It was a rant. That said. we have tons of things like `whatever =  new wxBitmap(...)`. Which forces to handle the deallocation ourselves. If we made by value instead ``whatever =  wxBitmap(...)`  wxWidgets handles allocation/deallocation. And since the object is re-counted it is as efficient.

We should not continue this discussion here, my bad. Listening in zulip if need be.

bdbcat, 2023-04-20:
Corrected by https://github.com/OpenCPN/OpenCPN/commit/d726364e3a0f83e64a0a936716f3a262ea3a57c4
Root cause:  incorrectly managed reference to object (wxBitmap) owned by plugin.

closed 2023-04-21 by nohal