OpenCPN/OpenCPN #3155: O failure  related to navobjects.xml.changes and bad plugins

rgleason, 2023-04-21:
**Describe the bug**
Start Opencpn 5.8 + current release
and it fails immediately.  Cleaned out some of the plugins in the cache because I believe objsearch turned all the icons into green puzzles and I am prompted for "safe mode" which I select "yes" usually. (In trying to fix the green Icons issue, Opencpn started failing immediately.)

I find in the log the following line.

```
07:21:31.432 MESSAGE navutil.cpp:1704 Done loading navobjects, 0 duplicate waypoints ignored
07:21:31.432   ERROR filefn.cpp:1045 File 'C:\ProgramData\opencpn\navobj.xml.changes' couldn't be removed (error 32: The process cannot access the file because it is being used by another process.)
```
This file exists in C:\ProgramData\opencpn  and it appears to be empty.
I deleted it.  

Restarted Opencpn and all plugins were still green puzzles.  Closed.
Deleted all cache metadata and tarballs. Deleted any dlls in %localappdata%\opencpn\plugins that were suspicious. 

Restarted Opencpn and the plugins no longer were green puzzles and O no longer failed.
I think that objsearch or some other plugin may be the culprit insofar as the bad plugin, but not sure.
I don't know why I was getting the navobj.xml.changes issue.

**Desktop (please complete the following information if applicable):**
 - OS: Win 11   Opencpn 5.8 Release

bdbcat, 2023-04-21:
Rick...
There is maybe one sure way to resolve this kind of problem.
1.  Start with clean (really clean) installation of O562, installed as parallel  All caches empty, AppData\Local\opencpn\plugins empty, etc.
2. Start O562.
3. Load and enable a managed plugin.
4. Stop O
5. Start O580.
6. look for crash.
7. If no crash, stop O580, take notes, choose another plugin, and go to step 2.

Tedious, but should be effective in identifying any managed plugin that causes trouble on upgrade..

Dave

rgleason, 2023-04-23:
Steps taken (I don't want to loose some user data for testing):

- C:\Users\fcgle\AppData\Local\opencpn   renamed plugins to plugins-OLD and recreated plugins folder.
- C:\ProgramData\opencpn renamed plugins to plugins-OLD and recreated plugins folder.
- Start O562   (in order to get O562 to work without failing, I had to create the "cache" folder.)
- Now Options isn't working and I have to use taskmanager to close O562

Not done yet
- Load and enable a plugin. Close O.
- Start O58
- Look for crash.  Close O.
- Start O562 and repeat.

bdbcat, 2023-04-23:
OK, following.  Thanks

rgleason, 2023-04-23:
See above.
- Now what?  Opening O58 won't test the plugins.

rgleason, 2023-04-23:
I suppose I could try deleting the opencpn.ini each time I switch programs, bu then the enabled plugins would be lost.  Perhaps just keep those?

rgleason, 2023-04-23:
To get O562 working normally without portable mode I tried this:

- Deletion of opencpn.ini,  delete opencpn.log (for clarity), 
- removal of  everything in C:\Users\fcgle\AppData\Local\opencpn\plugins
- removal of everything in C:\ProgramData\opencpn\plugins\cache\metadata
- removal of everything in C:\ProgramData\opencpn\plugins\cache\tarballs
- removal of everything in C:\ProgramData\opencpn\plugins\install_data
- removal of C:\ProgramData\opencpn\plugins\weatherfax  downloaded fax data

Is it possible that the C:\Users\fcgle\AppData\Local\opencpn\share\locale  data is screwing this up?
[0dca0d4c-3739-4d6e-b9ce-568c94aba88e.zip](https://github.com/OpenCPN/OpenCPN/files/11304383/0dca0d4c-3739-4d6e-b9ce-568c94aba88e.zip)

rgleason, 2023-04-23:
It is a wxWidgets exception.
Is there a better version of 5.6.2 that I should be using?

bdbcat, 2023-04-23:
Rick...
Following along, duplicating most of the process.
The trouble is in two cases:
1.  Trying to revert to O562 after having run O580.
2. Trouble with O580 using some older plugins.

Taking these in turn.
1.  To get O562 to run after running O580, the problem seems to be with C:\ProgramData\opencpn\iconCache.  On running O580, it gets filled with a huge number of files, which seems to choke O562 later.  Preliminary thought, not yet convinced.
If you delete  (or rename) this folder, O562 will start OK for me.

2.  O580 with bad older plugins:  Not worked on this yet.

rgleason, 2023-04-24:
C:\ProgramData\opencpn\iconCache  contains about 14,000 svg files which have the full pathname with "_" separator of every version of opencpn 5.7.1 + XXXXX that was ever installed on the machine.  I have removed all svg icons starting with C_ProgramFiles (x86) ....   which is about 15,000.  

Thanks, that should help.

bdbcat, 2023-04-24:
Yes, after some discussion, we have a plan.  I will soon post a patch for O580 that uses a different location for its iconCache, so as not to interfere with O562's cache location.

rgleason, 2023-04-24:
I removed all those 5.7.1 icons svg and was able to do a full "round trip" with weatherfax which passed with flying colors. Each time there was a message to reinstalled the plugin because it was not compatible.    Remove extraneous iconCache svg, then
O562 --> Wxfax message--> O58 ---> Wxfax message ---> O562

rgleason, 2023-04-24:
In C:\ProgramData\opencpn\iconCache

I found another 365 svg files added from the 5.8 installation.
I have removed them again, they come from 

- C_Program Files (x86) ....
- C_Users_fcgle_AppData_Local_opencpn_plugins_weatherfax_pi.svg  etc.

rgleason, 2023-04-24:
So many files just overwhelm the system. Why the need for the full path?

bdbcat, 2023-04-24:
Let us keep in mind that this is something of a "special case".  Reversion of O58x to O562 should likely be a rare event in the real user base.  But it could happen.  So we can prevent problems for future users by this change.

rgleason, 2023-04-24:
Thanks so much Dave. Will delete iconCache and get current with master! 
And will test all plugins I can lay my hands on between 0562 058 and back to 056
and make a list.

I think I'll make a new issue under Opencpn/plugins.  
Closing this now.

closed 2023-04-24 by rgleason