OpenCPN/OpenCPN #3157: Plugin Manager does not support OCPN portable mode.

bdbcat, 2023-04-21:
OCPN 5.8.x is coded to search for managed plugins in a fixed location.  On Windows, this is something like:

`C:\Users\dsr\AppData\Local\opencpn\plugins`

The common OCPN portable use case calls for the entire runtime environment, with all dependencies and plugins included, be  combined into a single image suitable for transport on, for example, a USB thumb drive.

The default plugin location mentioned cannot be added to a portable OCPN image with requiring its contents to be copied to the target system.  This breaks the "portable" concept.

Let's discuss.

rgleason, 2023-04-21:
I think the windows location you identified would be fine. I also think [Terry yachtvalhalla wou](https://www.cruisersforum.com/forums/f134/why-climatology-files-in-program-data-274831.html)ld appreciate that location.

leamas, 2023-04-21:
This is basically  #2233, or part of it. Some bits and pieces:
  - On Linux, the path to load from can be changed using environment variables, notably  XDG_DATA_DIRS. It would be simple to add something similar to Windows. More flexibility, but requires reasonably skilled users.
  - Here is also #1022, using an alternative configuration directory. Many  blocking  issues in this bug are now fixed after the plugin manager -- this has enforced that plugins uses the correct way to find data locations etc.
  - We really need to define portable mode vs the plugin manager. This has not been done..
  - Using an alternative path for loading plugins while still keeping the same configuration directory will give interesting results since the installation data in the configuration dir does not match the actually loaded plugins.

rgleason, 2023-04-21:
Well, Alec, I have no problem if the portable problem is fixed by using the new standard external plugin location!!!!   ...for windows this is, as you showed us,  %localappdata%\opencpn\plugins.
I have been telling Terry to just move plugin DLL and data directories to that location, but he doesn't think it will work.

leamas, 2023-04-21:
> I have been telling Terry to just move plugin DLL and data directories to that location, but he doesn't think it will work.

It will certainly work, but then he needs to set LOCALAPPDATA to  the actual directory to be used. 

Windows environment variables are not case sensitive, but documented using ALL CAPS -- this is the general convention for environment variables.

Hakansv, 2023-04-21:
>  A possible location might be, on WIndows, something like
> {portable_root}/ProgramData/opencpn/ManagedPlugins/
> All managed plugin manipulation and access would use this "portable plugin" directory.

The "old fashion way" to make a portable instance was to: Copy c:\program files\opencpn\*.* to any location, i.e. a USB stick.
(Since all plugins was in the same directory this was not a problem.)
Start it with e.g. d:\opencpn\opencpn -p and the log and config files is created in "d:\opencpn". No problem.
Now managed plugins and corresponding files, have to be installed on d:\ instead. We could do this anywhere on d:\ like the path mentioned by Dave or maybe: "{portable_root}/opencpn/ManagedPlugins/ to keep all files below "/opencpn/"?
The same pattern for default charts installation.
For Windows no permission issues whatsoever. The same for Linux/Mac I suppose?
We would not expect the user to make more copying than the initial program files?

bdbcat, 2023-04-22:
Let's review the requirements for a portable image, from both the packager and the user's perspectives.
For simplicity, I consider Windows only here.  By "image", I refer to a directory containing the portable OCPN instance, and all required parts.  Call it _{portable_root}_.

1.  Image must be complete, containing all dependent dll's, except for those expected to be on the target system already.
2.  Image may, if desired, contain pre-installed managed plugins.
3. There must be no formal "installation" process required to use the image.  No Admin mode access, nor adjusting of system environment values, either at system or user level.
4. On OCPN exit, and removal of the image media, or deletion of the image, (if copied), there must be no leftover traces of OCPN on the target system.

This may be an aggressive spec.  But it certainly serves the user requirement that we have observed in previous OCPN versions.
Comments?

bdbcat, 2023-04-22:
After looking into the OCPN580 code base, and some experimentation, I find that we can implement the spec above for Windows, at least.
Assuming the opencpn executable is:

`{portable_root}/opencpn.exe`

Then the proposed solution requires a one-line source code change, allowing the directory
` {portable_root}/plugins`
to be used for managed plugin storage.  This works for both pre-installed plugins, as well as run-time managed plugin download and installation.  No environment change is necessary.

transmitterdan, 2023-04-22:
> After looking into the OCPN580 code base, and some experimentation, I find that we can implement the spec above for Windows, at least. Assuming the opencpn executable is:
> 
> `{portable_root}/opencpn.exe`
> 
> Then the proposed solution requires a one-line source code change, allowing the directory ` {portable_root}/plugins/` to be used for managed plugin storage in portable mode. This works for both pre-installed plugins, as well as run-time managed plugin download and installation. No environment change is necessary.
> 
> Comments?

This is the way.

Hakansv, 2023-04-22:
I fully concur ...

leamas, 2023-04-22:
I have ended up in something similar, partly from another angle:  if OpenCPN actually behaved like a Windows application should behave, there is actually no need for a portable option (sic!)

Well-behaved windows applications are supposed to store their data in APPDATA i. e., the value of %LOCALAPPDATA% if i exists, otherwise a location under %USER%. This is now used for plugins only.  A new strategy would then be on windows:
  - Moving everything  now under ProgramData to APPDATA
  - Load "portable" data from  APPDATA
  - Keep plugins as-is

Likewise, on Linux we would move the configuration from ~/.opencpn to the standard path CONFIG_DIR i. e., the value of $XDG_CONFIG_DIR if it exists, otherwise ~/.config/opencpn. I hav'nt dived into MacOS, assuming something similar is possible also there.

In the end, this would mean that we would comply with the overall ideas of how applications should store their data on  different platforms. This is actually useful: the data handling we did for the plugin manager worked out of the box when when  we implemented flatpak. Just an example.

This would also mean that there is no need for a portable option. The portable use case would be about copying a complete folder and setting %LOCALAPPDATA% on windows, XDG_DATA_DIR ocn Linux. 

And, important, it would avoid the problems which inevitably will come if we let the configuration directory (which also contains the installer data) come "out of sync" with the actual plugins installed. 

Of course there are questions how to make such changes without breaking existing installations. However, while it requires some thought I don't think it's  that  complicated.

transmitterdan, 2023-04-22:
The use cases for portable mode don't care about the OS paradigm:

Here is one use case for portable mode:

1. Copy my portable O folder to a USB drive.
2. Then take that USB drive to an on-shore public computer and update all my charts, gribs, plugins, etc. using that computer's wi-fi connection.
3. Take the USB drive back to my boat, plug it in and copy it back over my O portable folder.

(Yes, I know this could catch me a virus, but for now I don't care about that.)

Another use case is to be able to run multiple instances of O simultaneously.  Granted, we could redesign O to allow multiple instances, but today it's not possible using a normal installation, at least on Windows.

In my mind, separating the program from its data, defeats the purpose of portable mode.  Everything (executables, data, charts, plugins, downloaded tarballs, tide/currents, etc.) has to be under one common folder tree to make portable mode viable.

leamas, 2023-04-22:
And that would work just fine with "my" approach as well, unless I missed something. Most of the stuff is in the configuration directory, and would hence be part of it automatically.

Charts, however, is a different beast and would probably need special attention unless they are stored in the default directory... hav'nt thought about them. But it's a problem in any case due to the sheer size.

My point is basically that a lot of pieces falls in place if the "portable folder" is the same as the "configuration directory". I also envision that we have the dlls available there. Since the directory would be defined by %LOCALAPPDATA%  the main executable could be anywhere, including on a stick.

In other words, it would not be about separating the program from it's data. On the contrary, it would be about keeping everything needed in one well.defined place not only in portable mode but also in the general case.

bdbcat, 2023-04-22:
I agree that more thought on this would be productive for O5.10.x.  I am not attracted by the idea of requiring environment changes at run-time.  This is not a "normal" thing for Windows users.

In the moment, though, for O582, I think that the minimally intrusive plan will suffice for Windows, which is our primary use case.  The patch is in master now, so it could be tested.  @Rick, perhaps you could bring Terry up to speed and monitor his testing?

rgleason, 2023-04-22:
Dave, I'll git pull in 3 hrs and try it.  Will also start testing PI transition of PI between o562 and 058 next.
I find Alec's %LOCALAPPDATA% suggestion a good one, even with and ENV change, but perhaps that is for 5.10.

bdbcat, 2023-04-22:
Rick...
The build is already available in Cloudsmith:
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.8.0-0+411.9e5f7f5_setup.exe

leamas, 2023-04-22:
Ok, let's have this bug targeting the minimal fix, leabing #2233 to handle more thoughts and perhaps bigger changes.

leamas, 2023-04-22:
Btw: using portable mode requires using command line options, which isn't much better than environment variables on windows...

transmitterdan, 2023-04-22:
Dave,

I don't have time today to debug this (I'm on the boat). But nothing has changed for me. Plugin-manager still refuses to install a plugin in portable mode. It says it is installed but it really isn't. AFAICT it downloads the plugin, finds it compatible with wx and what not then proceeds to destroy the plugin .DLL because it's found in the plugins folder which it thinks is wrong.

nohal, 2023-04-22:
The commit seems not to be in `master`, but only in the `glew_bionic` branch for now

Hakansv, 2023-04-22:
> Btw: using portable mode requires using command line options, 

No. On Win you just make a shortcut to opencpn.exe and edit it by adding the "-p", like this: 
"C:\Portable\OpenCPN\opencpn.exe" -p

transmitterdan, 2023-04-22:
> The commit seems not to be in `master`, but only in the `glew_bionic` branch for now

Doh!

bdbcat, 2023-04-22:
Ooops.
Building now....
Done.
https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/opencpn_5.8.0-0+415.1b87b53_setup.exe

The relevant commit is:
3b3e305e5047f57650ac5b52bb405a0456d2fce9

Hakansv, 2023-04-22:
Tested on Win10.
Works like charm.
I updated master catalog and installed o-charts. It was installed in C:\ProgramData\OCPN_Port\plugins. (That's where I've run portable for years. I know it's not a recommended location due to permissions but the goal was to test installation of a managed plugin.)
Since the charts used are the same as for the standard O on this machine all was working.
For info I also attache my old bat-file from the time before dual canvas when I used to run a portable instance all time. Still works.
Shot from tipple canvas.. :-)

[image]

[UpdatePortable.bat.txt](https://github.com/OpenCPN/OpenCPN/files/11302248/UpdatePortable.bat.txt)

rgleason, 2023-04-22:
Have downloaded, copied to D: memory drive and installed.  Opened from command prompt with opencpn.exe -p
Updated PIM master catalog. Installed weatherfax and climatology and both work. The dlls  and data directories all land in the "plugins" directory on the D: drive under Opencpn.     (including climatology_pi/ usericons and data directories, unlike the standard installation)

[image]

Created a new folder "Charts" under OpenCPN on the D: drive.  Used Chartdownloader to download the USCG01 Region. and tried it out.  Works well.  There is one small yellow rollover box with very small text that is a problem on this high res display.

[image]

All files are contained in the one folder D:\OpenCPN 5.8.0-0+1b87b53
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/11302336/opencpn.log)
[opencpn.ini.txt](https://github.com/OpenCPN/OpenCPN/files/11302337/opencpn.ini.txt)

The one suggestion I have is for the User to amend the opencpn.ini file at this line with a text editor:
BaseChartDir=C:\\Users\\,useraccountname>\\OneDrive\\Documents\\Charts

to be something like 
BaseChartDir=D:\OpenCPN 5.8.0-0+1b87b53\Charts

closed 2023-04-26 by bdbcat