OpenCPN/OpenCPN #3170: Failure to start with OpenGL enabled on VM without accelerated graphics

nohal, 2023-04-28:
I use qemu/kvm for virtualization. It does not support OpenGL in Windows guests. After installation we set `OpenGL=1` in the ini file, which leads to a pretty early crash. We should probably consider trying to disable OpenGL in case we last time crashed this early and it is enabled.

nohal, 2023-04-28:
I did not run the debugger on it, but from the log it seem pretty obvious that we crash in https://github.com/OpenCPN/OpenCPN/blob/master/src/glChartCanvas.cpp#LL1050C22-L1050C22

bdbcat, 2023-04-28:
This is addressed by #3152
We do need a better way to safely test OpenGL during launch, and is on my roadmap for next version.
For now, the commit referenced by #3152 allows complete disable of OpenGL by manual config file entry.  Trying to be clever about early crashing is, IMHO, risky for a "not-to-be-heavily-tested" maintenance release.

nohal, 2023-04-29:
Yes, disabling OpenGL in the config file works fine with the current builds, but is hardly something a normal user figures out. What about at least mentioning it in the safe mode dialog we display after the crash in 5.8?

nohal, 2023-04-29:
I can imagine adding a hyperlink there leading to a page dedicated to known problems leading to crashes and their solutions not to make the text itself grow indefinitely.

bdbcat, 2023-04-29:
Both good ideas for a future version.
But two points:
1.  This problem is exceedingly rare.  Your instance is only the second seen in the entire O58 test series.
2. We have serious problems today with O580 (cm93 with OpenGL on Windows is simply fatally broken).  I'd rather not need to go through a crowdin cycle taking at least a week to update the messages just now.

Maybe we will do O584 in the summer?

nohal, 2023-04-29:
Yes, I agree this will be a very rare sight and we need 5.8.2 ASAP, no problem to postpone this. And no problem with 5.8.4 later this year (Probably not a bad idea to do with whatever appears by then plus an updated manual)

bdbcat, 2023-04-29:
OK, thanks.  I'll change the milestone.

bdbcat, 2023-04-29:
Or maybe you should.  My change did not take...

leamas, 2023-04-29:
IIRC, the same issue was discussed with @jongough some time ago.  I don't find the issue right now, but I think the conclusions `was:`
  - It is possible to work around the problem by installing mesa for windows. The resulting performance is bad due to software emulation, but it works.  I have done this, and talked about in the issue I don't find. And I have no access to my windows box over the weekend..
  - This is a a developer only setup. Real users runs on bare metal, and have OpenGL available. 
 
The conclusion in the previous case was that we should not really care about it, it's up to developers to fix their environment.

nohal, 2023-04-29:
I don't have a problem with the environment, don't need OpenGL there. I just think we could be better helping the users solving their startup issues. And a very cheap step in that direction is https://github.com/OpenCPN/OpenCPN/pull/3171

leamas, 2023-04-29:
> And a very cheap step in that direction is https://github.com/OpenCPN/OpenCPN/pull/3171

Agreed

rgleason, 2023-04-29:
Perhaps this is the thread with Jon Gough
https://github.com/OpenCPN/OpenCPN/issues/2911

nohal, 2023-05-18:
Fixed.

closed 2023-05-18 by nohal