OpenCPN/OpenCPN #3176: User icons scaling problem

nohal, 2023-04-30:
When user icon is first used, it is drawn small (Size of the icon in the dropdown)
[image]

After next start of the application and on all subsequent uses after that it is shown full size
[image]

Happens for both SVG and PNG user icons

nohal, 2023-04-30:
It is actually the case for all the icons, not just user defined ones.

bdbcat, 2023-04-30:
related to #3177?

nohal, 2023-04-30:
No, #3177 I just accidently found while looking into this.

nohal, 2023-04-30:
This one can probably wait for next release, but https://github.com/OpenCPN/OpenCPN/pull/3178 which is fixing #3177 should go to 5.8.2

bdbcat, 2023-04-30:
"for all the icons," ?
Strange we never saw this in test.
Quick test on Windows, no problem.
Where is the icon shown come from?  Built with Inkscape, or ??

nohal, 2023-04-30:
Yes, visible well on the sea bottom icons

When first selected
[image]
On next start or after switching color scheme
[image]

bdbcat, 2023-04-30:
OK, I see that on Win.  Will investigate.  I'll hold #3178 until we know more.  Could be related

nohal, 2023-04-30:
#3178 is not related, it is a real issue when the `svg` element does not have `width` and `height` attributes set - we then do not touch the values of the referenced variables in `SVGDocumentPixelSize` and if they were bogus before and we do not check for it after, ugly stuff may happen. That needs to be addressed as it may be invoked by the plugins asking for a bitmap from their SVG so we don't have control of that.

bdbcat, 2023-04-30:
Agree.  A serious bug.  Will definitely be in 582.

bdbcat, 2023-04-30:
I added a fix for this, but I do not exactly trust it for the general case.  Needs wider testing and leak check.
Let's leave this open for 5.10.x

nohal, 2023-04-30:
No problem, this really is not a big deal

nohal, 2023-04-30:
The code in https://github.com/OpenCPN/OpenCPN/commit/2d30b87abf043345bde39a0140a5cfe27eb0011a does not seem to do the trick for me.

bdbcat, 2023-04-30:
Sigh...
Leave it for now.

bdbcat, 2023-05-01:
btw, which Icon are you having trouble with?

nohal, 2023-05-01:
Not exactly an icon, just a random SVG I created and put into `UserIcons`. Attaching.
[image]

nohal, 2023-05-01:
BTW, the 100x200 size is there on purpose to illustrate another problem we have with the mark icons - making them square. We need some general refactoring here for 5.10...

bdbcat, 2023-05-01:
Yes, the refactor to separate GUI from CMD mode introduced a bunch of spaghetti.  I've been simply hacking it ever since. Needs work.

bdbcat, 2024-02-28:
Corrected sizing logic of arbitrary icons dropped into UserIcons directory, when used as Mark icons.

Not sure how to handle plugins.  My inclination is not not fiddle with icons loaded by a plugin.  I want to assume that each plugin knows what it wants, and does not expect it to be edited by core.  Maybe simplistic?

Also, brings up another issue:  We currently cache svg icons after initial rendering, for performance.  This seems OK for inbuilt svg icons, as shipped with release.  But we do not want to cache UserIcons, I think.  A user may need some time to get the icon exactly right to their preferences.  Trying several times.  If cached, the user would need to find and delete the cached copy.  Tedious.  Maybe want to do the same for all plugin svg operations?

nohal, 2024-02-28:
I would not touch this much more now. What we probably need is some metadata defining the behaviour of the icons - https://github.com/OpenCPN/OpenCPN/issues/3193.

bdbcat, 2024-02-28:
Can we close this, or...

closed 2024-02-28 by nohal