OpenCPN/OpenCPN #3182: Implement support for online tiles with caching

nohal, 2023-05-01:
There is a growing number of interesting sources of map tiles available online. We already support MBTiles which builds on the same logic. Our current MBTiles implementation is performing all the work in the main thread and clearly can't scale for the several orders of magnitude slower online sources, so it needs to be made asynchronous to be able to be shared with the online sources. Some PoC already done in https://github.com/OpenCPN/OpenCPN/pull/1293

Various online tile services differ in the way they serve the data, the most basic looks like `https://tile.openstreetmap.org/{z}/{x}/{y}.png`, but some require authentication and additional logic to obtain the tiles.

Popular third-party tools already allow caching online tiles locally to disk and conversion between cache types.

Probably the most performant cache type able to scale to huge tilesets uses a directory tree with the following structure

`<cache root>/<cache name>/z<zoom level>/<x >> 10>/<y >> 10><x >> 8>.<y >> 8>.sqlitedb`

Where the database looks like

```sql
create table t
(
    x INTEGER           not null, -- x tile coordinate
    y INTEGER           not null, -- y tile coordinate
    v INTEGER default 0 not null, -- ??? (always 0?)
    c TEXT, -- content type if different from the main content type of the cache
    s INTEGER default 0 not null, -- original tile size in bytes
    h INTEGER default 0 not null, -- CRC32 checksum
    d INTEGER           not null, -- unix timestamp of acquisition of the tile
    b BLOB, -- tile image data
    constraint PK_TB
        primary key (x, y, v)
);

create index t_v_idx
    on t (v);
```

LeaseOnLife, 2023-05-02:
yeah! shoot me a note if you want help testing. Above would also 'kill' FS2495:
https://opencpn.org/flyspray/index.php?do=details&task_id=2495

itemir, 2023-12-20:
Strong +1. This would be a super awesome follow-up. There are a ton of map tiles available online and Internet becoming more ubiquitous on boats thanks to Starlink and other similar technologies, online map sources are becoming a lot more usable at least for cruise planning, if not for navigation.

itemir, 2025-02-14:
Has there been any progress to this? With Internet being more readily available, online chart support (like a fileserver) would be a significant enhancement for OpenCPN.