OpenCPN/OpenCPN #3186: Implement more AIS type 8 binary messages

nohal, 2023-05-02:
There are reports that more binary broadcast messages have been implemented and are used by various bodies.
One such report:
> I just got the info from the provider, it’s AIS ASM message 8, DAC 367, FL 33, it’s binary broadcast message that contains buoy or Aton weather observations, along with the lat/long position of the observation. The NMEA 2000 pgn is 129797

## Relevant documentation

https://www.e-navigation.nl/asm/ais
https://gpsd.gitlab.io/gpsd/AIVDM.html#_type_8_binary_broadcast_message
https://arundaleais.github.io/docs/ais/ais_decoder_binary.html

akrovers, 2023-05-04:
Hello! I just did a quick capture on one Message 8 transmission. I will get more.   !AIVDM,1,1,,B,8>k1oKiKpB9GO>42lBPWmre0<N00,0*55

akrovers, 2023-05-09:
Here is how the data is displayed on Rose Point Explorer 
[image]

nohal, 2023-05-16:
We can crosscheck against the new implementation in https://github.com/fulup-bzh/GeoGate/blob/master/encoder/lib/GG-AisDecode.js#L515-L565

Rasbats, 2023-07-21:
My alpha plugin aisMET_HYDRO_pi is capable of reading these messages:
[https://github.com/Rasbats/aisMET_HYDRO_pi](https://github.com/Rasbats/aisMET_HYDRO_pi)
Tarballs are here:
[https://cloudsmith.io/~opencpn/repos/aismet_hydro-alpha/packages/](https://cloudsmith.io/~opencpn/repos/aismet_hydro-alpha/packages/)
Number of stations are limited. Here is a screenshot of stations found so far:

[image]

Hakansv, 2023-10-23:
I do agree about the intentions in this issue.
One example:
The Meteorological and Hydrographic data are available in message 8 DAC = 001 FI = 31
We actually have a bug in OCPN where these messages shows up as Class A target for a second and disappears thereafter. It's traceable and it looks not good.. See pict 1. 
Since I recognized these "Class A" targets I started to investigate and found out the cause. 
I also tried to catch them and using IMO SN.1/Circ.289. I was able to decode the Swedish west-coast broadcasts from the base stations around.  See a code snippet and debug examples below.
It's valuable information for most users and I think we would implement it in OCPN.

This implementation can be in the core code or as a plugin as of rasbat's example above. (I've made a short test and will comment in next post.)
For the time being I prefer we implement it in OCPN. The information may be handled as a new AIS class "AIS_METEO"? This may produce an icon for each received station. Preferable like an ATON but a circle big enough to not cover the transmitter buoy or beacon. The object query for the meteo target would list available meteo data.

I could decode the message 8, DAC 001, FI 31. Not a big problem except the minus sign as per 2's complement due to my lack of  bit handling knowledge.
The next problem will be to produce a AIS target for the different stations.
We can't simply use the AisTargetData base using a new AIS class "AIS_METEO" since that data base use the MMSI number to identify each station/target.
The meteo stations here use the same mmsi number for all station on the west coat and another mmsi for all stations on the east coast.  See my debug list herein where it's clear all stations use the same (00)2655619 Message ID.
So we must divide and identify each station using the lat/lon position. For that I think we need a new target array making the implementation complicated. Thoughts about this more than welcome.

I used this document to decode the AIS message 8. [IMO SN.1/Circ.289 See Annex A](file:///C:/Users/HAS-WIN/Downloads/TWLWG7-4.2.3-TSMAD29_DIPWG7-06.4C_Draft_S-112_DynamicWaterLevelDataProductSpecification.pdf)
I also made me a "quick reference guide". Attached.

Decoding is though not a big issue. A new array to identify each received station based on lat/lon is more tricky as I sees it.

Discussion please.

Thanks
Håkan

Pict 1
[image]

Code snippet
```
ais_decoder.cpp: 3182
if (fi == 31) {  // Meteorological and Hydrographic data
..................
met.day = bstr->GetInt(107, 5);
met.hour = bstr->GetInt(112, 5);
met.minute = bstr->GetInt(117, 6);
msgFi += wxString::Format(" Date Time: %d %d:%d", met.day,
                          met.hour, met.minute);

met.wind_kn = bstr->GetInt(123, 7);
met.wind_G_kn = bstr->GetInt(130, 7);
met.wind_dir = bstr->GetInt(137, 9);

if (met.wind_kn < 127) {
  validmeteodata = true;
  msgFi += wxString::Format("   Wind speed: %d Kn  direction: %d\u00b0",
                            met.wind_kn, met.wind_dir);
}

int tmp = bstr->GetInt(155, 11);
met.air_temp = tmp / 10 + (double(tmp % 10))/10.0;
// TODO Handle negative temp as per 2's complement
if (tmp != 1024) {
  validmeteodata = true;
  msgFi +=
      wxString::Format("   Air temp: %.1f \u00b0C", met.air_temp);
}

// Relative Humidity
// Dew Point
// Air Pressure
// Air Pressure
// Tendency
// Horizontal Visibility
// Water level(incl.tide)
// Water Level Trend

// Surface Current speed incl tide
int cur = bstr->GetInt(216, 8);
met.current = cur / 10 + (double(cur % 10))/10;
met.currDir = bstr->GetInt(224, 9);
if (cur < 255) {
  validmeteodata = true;
  msgFi += wxString::Format(
      "   Current speed: %.1f kn direction: %d\u00b0",
      met.current, met.currDir);
}

int wave = bstr->GetInt(277, 8);
met.wave_hight = wave / 10 + (double(wave % 10))/10;
met.wave_period = bstr->GetInt(285, 6);
met.wave_dir = bstr->GetInt(291, 9);
if (wave < 255) {
  validmeteodata = true;
  msgFi += wxString::Format(
      "   Wave Height: %.1f m period: %d s direction: "
      "%d\u00b0",
      met.wave_hight, met.wave_period, met.wave_dir);
}
```
Debug examples from the parsing above:
```
23:03:14.447 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 58.489100   Longitude : 10.930600 Date Time: 22 21:2   Air temp: 9.4 °C   Wave Height: 0.7 m period: 3 s direction: 76°
23:03:22.442 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 58.489100   Longitude : 10.930600 Date Time: 22 21:2   Air temp: 9.4 °C   Wave Height: 0.7 m period: 3 s direction: 76°
23:04:04.218 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 56.041167   Longitude : 12.684500 Date Time: 22 21:2   Wind speed: 9 Kn  direction: 225°
23:04:31.230 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 57.249800   Longitude : 12.112550 Date Time: 22 21:2   Wind speed: 6 Kn  direction: 132°
23:04:32.467 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 57.966900   Longitude : 11.728300 Date Time: 22 21:2   Wind speed: 7 Kn  direction: 77°
23:04:45.159 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 58.298350   Longitude : 11.327450 Date Time: 22 21:2   Wind speed: 14 Kn  direction: 76°
23:04:52.492 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 55.589367   Longitude : 12.844467 Date Time: 22 21:2   Current speed: 2.0 kn direction: 360°
23:04:56.666 MESSAGE ais_decoder.cpp:3286 **********Meteo station:  002655619
Latitude  : 58.350317   Longitude : 11.426467 Date Time: 22 21:2   Wind speed: 7 Kn  direction: 98°
```
[meteo_data_decodes.pdf](https://github.com/OpenCPN/OpenCPN/files/13074808/meteo_data_decodes.pdf)

Hakansv, 2023-10-23:
@Rasbats 
I tested your nice plugin on Win10
The result is one icon although there are at least ten available. 
Also no data meteo data visible on the icon. 
I suspect the trouble depends on all stations here use the same Message ID, MMSI? See my text above.
Håkan
[image]

rgleason, 2023-10-23:
Looks very interesting.

Rasbats, 2023-10-24:
@Hakansv : The green icon is a waypoint. Right-click->Properties brings up the data.

Hakansv, 2023-10-24:
Yes, I tried that but no data more than a positive Ice. (Not likely)
And there should be a lot more stations but all using the same mmsi!
Try my data to compare: TCP address: signalk.stupan.se Port: 10112

[image]

Rasbats, 2023-10-24:
2s complement from Kurt Schwehr:

```
int AisBitset::ToInt(const size_t start, const size_t len)  const {
	assert(len <= 32);
	// TODO(schwehr): Prefer to use num_bits (includes pad) for checking bounds.
	assert(start + len <= num_chars * 6);

	assert(current_position == start);

	// Converting the sub-bitset to a signed number, per "Two's complement":
	// - If negative, invert all the bits, then add 1.
	bool is_positive = (len == 32 || !test(start));
	int result = 0;
	size_t end = start + len;
	for (size_t i = start; i < end; ++i) {
		result <<= 1;
		if (test(i) == is_positive)
			result |= 1;
	}
	current_position = end;
	return is_positive ? result : -(result + 1);
}
```

Rasbats, 2023-10-24:
Navigation symbol for environmental report (ENV):
[https://iho.int/uploads/user/About%20IHO/International_Organisations/ECDIS-ENC/English/SN_Circ243-Rev.2%20-%20Guidelines%20For%20The%20Presentation%20Of%20Navigation-Related%20Symbols%20Terms%20And%20Abbreviations.pdf](https://iho.int/uploads/user/About%20IHO/International_Organisations/ECDIS-ENC/English/SN_Circ243-Rev.2%20-%20Guidelines%20For%20The%20Presentation%20Of%20Navigation-Related%20Symbols%20Terms%20And%20Abbreviations.pdf)

Hakansv, 2023-10-24:
Rasbats..
Very good. So a ring will fulfill the standard. I may have doubts about printing the "W" inside the ring since it will cover a possible buoy at the same position. Better above the ring?

Rasbats, 2023-10-24:
I was looking at the diamond with ENV inside!

Rasbats, 2023-10-25:
v0.3.0 uses a station ident of MMSI_lat_lon for AIS8_1_31. All the Swedish stations are showing, I hope.
 
[https://cloudsmith.io/~opencpn/repos/aismet_hydro-alpha/packages/](https://cloudsmith.io/~opencpn/repos/aismet_hydro-alpha/packages/)

Hakansv, 2023-10-25:
Mike..
Very good. You've solved most basics. What I can judge; nice code!
We can of course discuss a lot more details about presentation or so. But first one without having looked into all coding:
The lat/lon identification works fine here. See pict.
Q: Do you use the same resolution when comparing to recognize a single station as in the WP name?  Some of the stations are buoys that can move several meters by wind and current so we better don't use closer than 100 meters resolution.

[image]

Rasbats, 2023-10-25:
>Do you use the same resolution when comparing to recognize a single station as in the WP name?

Good point, yes the hashmap uses a station identifier with the resolution of the WP name. The values used to plot the waypoint are those reported by the station. If we used lat/lon with 1 decimal place for the identifier i.e. accuracy of 185m would there be stations that close together? It may not matter as one of the stations would be eliminated.

Vagspel, 2023-10-25:
Hi,
I am trying Hydro 0.3 on wx-64-10.
I can see in the ais list that it is getting info from 002655619 so OpenCPN is working so far.
It does not show any stations.
There are no received sentences.
When clicking on remove Hydro way-points OpenCPN crashes.
I am using data from Håkans server that is transferred from my AIS station. 
[image]
//Gunnar

Rasbats, 2023-10-25:
Not having problems here on a clean install of OpenCPN/aismet_hydro. After clearing waypoints I needed to toggle the button to green to start receiving again.

Perhaps try a direct link with TCP signalk.stupan.se on port 10112?

Vagspel, 2023-10-25:
Yes! It works with TCP. how do I read the data?
The new info on rightclick "RIS signal status" what is that?
//Gunnar

Vagspel, 2023-10-25:
Found data, "WP properties" rtfm :-)
//Gunnar

Hakansv, 2023-10-25:
@Rasbats 

> .. accuracy of 185m would there be stations that close together?

185 m would be fine. Non are closer that I suppose. 
Thanks

Hakansv, 2023-11-07:
I'm testing implementation into O core. Some examples of mouse rollover and Target query info below.
My intention is to make these Meteo points appearing when received. (Normally from  a base station)

I don't think we need an option, in "Ships->AIS Targets", to be able to switch them off? Some rings on chart won't disturb?
(Also are some of them already printed by chart data.)
Opinions please...

[image]
[image]
[image]

rgleason, 2023-11-12:
Very nice features. Thank you. 
Not quite the same, but squiddio has NOAA bouy & ship weather reports and AIS markers.

[image]

 
[image]

closed 2024-02-13 by nohal