OpenCPN/OpenCPN #3194: Shapefile support

nohal, 2023-05-03:
A lot of datasets that would be useful as overlays on charts are available as shapefiles.
Also there are high resolution land polygon and coastline datasets available from https://osmdata.openstreetmap.de/data/land-polygons.html and https://osmdata.openstreetmap.de/data/coastlines.html that could perhaps replace the aging GSHHG basemap.

loconox, 2023-05-29:
In France Shom provides opendata for beacons as shapefile : https://diffusion.shom.fr/donnees/balisage_maritime.html

bc38, 2023-05-29:
> In France Shom provides opendata for beacons as shapefile : https://diffusion.shom.fr/donnees/balisage_maritime.html

Still, it's more for GIS where you can load styles.
It should be easier for CPN when loading plans, diagrams, individual OSM datasets

loconox, 2023-05-29:
I'm not aware of technical limitations, but Qtvlm allows to load shapefiles.