OpenCPN/OpenCPN #3196: Display broken display. Current master (2023 May 04)

Ptizef, 2023-05-04:
On my Surface GO 2 W11 22H.
New built from current master, the display is partly broken. 
In the shot attached, you can see the broken display of the hamburger menu and the toolbar.
Also, all the texts on the chart but also in the dialog are excessively bold. Plugins are also concerned.
This happens since the "Update the Windows application manifest to support new DPI awareness features" commit.
[image]

Ptizef, 2023-05-04:
Add : 
Also, the status bar no longer apply the font size changes

Hakansv, 2023-05-04:
Although a sandbox(master) issue may not yet be called a bug I see similar effects on Win10.
Also most detailed scaling, e.g. ship scale factor and Dashboard, has to be redone but once made the same settings make sense. E.g. AIS targets was first to big but setting the scale factor to -2, Apply, and back to previous -1, Apply, did restore to desired size.

[image]

Hakansv, 2023-05-04:
Connected to display or not---
"Navigate to here" Is activating the route created but it's "deleted" after one or two seconds.

Hakansv, 2023-05-04:
Yes, sorry. I first thought it was a disturbed or hidden display but it may be the function itself.

nohal, 2023-05-04:
Even if it is only a problem with showing the console window, please open a new issue for it.

Ptizef, 2023-05-08:
I would suggest this simple change to retrieve a "normal" display

app.manifest
-            <gdiScaling xmlns="http://schemas.microsoft.com/SMI/2017/WindowsSettings">true</gdiScaling> 

+            <gdiScaling xmlns="http://schemas.microsoft.com/SMI/2017/WindowsSettings">true/pm</gdiScaling>

nohal, 2023-05-08:
Can you point me to the source of this information? All I can find is https://docs.microsoft.com/en-us/windows/win32/sbscs/application-manifests#gdiscaling and it does not mention `true/pm` as being a valid option for `gdiScaling` (it is valid for `dpiAware` where we do use it). I do really want to make this right this time, not just revive our old hacks by implementing more of them...

Ptizef, 2023-05-15:
Hi
some more tests 
gdiScaling active (current master)
my screen is 1920x1280
User Windows screen scaling parameter:
100%                              getDisplaySize() returns 1920 x 1280
175% (my preference)    getDisplaySize() returns 1097 x 731
gdiScaling deactivated
my screen is 1920x1280
User Windows screen scaling:
100%   getDisplaySize() returns 1920 x 1280
175%   getDisplaySize() returns 1920 x 1280

So gdiScaling reduces the screen definition. That is why the display seems a bit blurry at my preferred 175% parameter.(not to mention the broken displays shown above, which can be corrected)
JP

Hakansv, 2023-05-16:
Although a single monitor but the disturbed Toolbar and (Hamburger)menu are now very fine on W10.
Scaling in general is also improved. 
Thanks

Ptizef, 2023-05-16:
With single monitor and W11
Everything's fine now.
Unfortunately, I have no equipment to test multi-monitor.
Thanks.
 JP

bdbcat, 2023-05-16:
Backport to O58x now?

bdbcat, 2023-05-16:
OK, fine.  We'll hold it for later.

closed 2024-02-13 by nohal