OpenCPN/OpenCPN #3199: Translations:  debian packaging warnings.

leamas, 2023-05-05:
**Describe the bug**
Translations warnings from debian packaging
```
gettext: po/opencpn_hi_IN.po:7300: 'msgid' and 'msgstr' entries do not both end with '\n'
msgfmt: found 1 fatal error
gettext: po/opencpn_th_TH.po:41: 'msgid' and 'msgstr' entries do not both begin with '\n'
msgfmt: found 2 fatal errors
```

bdbcat, 2023-05-07:
This happens when the .po files have been manually updated, or a new translation downloaded from crowdin which does not match the '\n' format of the original.
Solution:  Manually edit the .po file, on the indicated line, and fix the new line characters as suggested.
Curious how you got this error, since I have not merged the translation files since 582 was released.
Any insight?

leamas, 2023-05-07:
Not really, I have generated the package from the release tag, and while there are some patches none of them affects the po/ directory.

However, I this might very well have been present also in 5.8.0, I did not check the [developer information](https://tracker.debian.org/pkg/opencpn) for this package (I should have, but missed it).

leamas, 2023-05-07:
Note the command used to find these:  `  msgfmt -c -o /dev/null <po file>` . It would be easy to add a cmake target which made this check for all po files which could be used at least prior to releases.

closed 2023-05-07 by bdbcat

bdbcat, 2023-05-07:
Fussing with '\n' mismatch is a normal part of my pre-release workflow.  crowdin could be more helpful here, since new-line characters are not so easy to spot in the crowdin translation GUI.  Translators do their best, but always some work needed whenever I download a new translation set.

leamas, 2023-05-11:
@bdbcat : 

Here is a patch which adds a new "po-check" target which can be used to display warnings like the ones reported in this issue.  Merge if useful.
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 1c74a7426..9fdd7ee22 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2544,6 +2544,13 @@ if (GETTEXT_MSGFMT_EXECUTABLE)
     add_dependencies(${PACKAGE_NAME} i18n)
   endif ()
 
+  add_custom_target(po-check)
+  add_custom_command(
+    TARGET po-check
+    COMMAND for f in *.po\; do msgfmt -c -o /dev/null $$f\; done
+    WORKING_DIRECTORY ${CMAKE_SOURCE_DIR}/po
+  )
+

```

aviatorhh, 2023-05-11:
To give my 2c, I tried to apply to the release package 5.8.2 but it seems to place it wrongly. Could you mention the source version for this patch? (Or I did something wrong)

> @bdbcat :
>
> Here is a patch which adds a new "po-check" target which can be used to display warnings like the ones reported in this issue. Merge if useful.
>
> ```
> diff --git a/CMakeLists.txt [...]

leamas, 2023-05-11:
It is against current master (f178779). 

It should be easy to apply manually also to 5.8.2, it's just some added lines before  ` endif (GETTEXT_MSGFMT_EXECUTABLE)`

PR #3202 (2023-05-07): po: Fix msgfmt warnings
Fix warnings described in #3199 

Closes: #3199