OpenCPN/OpenCPN #3200: When activating a route it's instantly deactivated.

Hakansv, 2023-05-06:
**Describe the bug**
Present master branch: 
When activating a route it's instantly deactivated.

**To Reproduce**
1. Create a route.
2. Activate it by the right click menu. 
3. The Console windows is shown.
4. Within some seconds the route is deactivated. Possible a false "completed"?
    The multiplexer did send the first messages to the AP though.

"Navigate to here" shows the same. Point: 3. and 4.

**Expected behavior**
The route would stay activated until manually deactivated or the ship's position is within the arrival radius of the last WP.

**Desktop 
 - OS: Win 10
 - Version Present master branch

**Additional context**
Last commit for this issue is: 81cfdf9335  (Housekeeping: some sources tidy up.)
Checkout commit:: f9ec2f982a4  (Merge pull request #3163...) and the issue is still valid.
Checkout commit: 5a198001c9 and this issue is not valid

leamas, 2023-05-06:
I'll take a look at this after the weekend.

leamas, 2023-05-09:
Trying to reproduce, but no luck.  Testing f17877939.

First test is to just add a route and activate it. For me, it stays active

Next step is to add live GPS data. I have done this on a Fedora/Flatpak instance with "real" data from a connected GPS.  Obviously, my desktop does not move that much. Route stays active.

 Trying a virtual Windows instance which get's the data from the Fedora/Flatpak over the network. Same result.

@Hakansv : Any hint?

leamas, 2023-05-09:
BTW: What happens if you activate a route without any navigation data at all? That is actually a quite important data point.

nohal, 2023-05-09:
@Hakansv `signalk.stupan.se:3000` seems OK to me, I am getting data from it right now.

Hakansv, 2023-05-09:
> signalK is unfortunately above my paygrade. 

paygrade - how come??  Cant you use a Signal k protocol type

> How long will it be before you are home and can fix the tcp server?

At least two weeks. Try this [VDR](https://www.dropbox.com/s/jc872vqze5okbqr/VDR_Hakefjord.txt?dl=1) instead if it's better.

> BTW: What happens if you activate a route without any navigation data at all? That is actually a quite important data point.

The same. Even wo NAV-data is the route deactivated instantly.

Hakansv, 2023-05-09:
Pavel: 
Yes, SK protocol port 3000 is OK. N0183 protocol port 10112 is down.

Alec:
It's not about a specific route. Any old/new as "Navigate to here" are the same.

leamas, 2023-05-09:
So, this does not depend on
  - route: it doesn't  matter  which use you use. , if any.
  - input data: it happens even if there is no data.

Configuration? Can you see it when using a (temporary!) fresh configuration?

Hakansv, 2023-05-09:
> Configuration? Can you see it when using a (temporary!) fresh configuration?

If you by "configuration" mean an empty config file so: Yes with one ENC chart set and no extra plugins it's still there.

Hakansv, 2023-05-09:
Pavel..
Routes may be your "baby"? Any hints where I should brake to debug what's happening?

leamas, 2023-05-09:
@Hakansv : have you any other idea what could be the difference between you and me (and Pavel). Could if be the build as such?

nohal, 2023-05-09:
> Pavel.. Routes may be your "baby"? Any hints where I should brake to debug what's happening?

Breakpoint at https://github.com/OpenCPN/OpenCPN/blob/master/include/routeman.h#L111 should allow you to see what got you there.

nohal, 2023-05-09:
And another one at https://github.com/OpenCPN/OpenCPN/blob/master/src/routeman.cpp#L435

Hakansv, 2023-05-09:
For any reason he walks in the code here (>>) and claim the route is ended.
```
void RoutemanGui::DoAdvance(void) {
  if (!m_routeman.ActivateNextPoint(m_routeman.pActiveRoute, false))  // at the end?
  {
    Route *pthis_route = m_routeman.pActiveRoute;
>>    m_routeman.DeactivateRoute(true);  // this
```

Hakansv, 2023-05-09:
I'm not a guru on this but he walks in here several times until m_arrival_test is counted to >2 and than arrival is true.
```
if ((m_routeman.CurrentRangeToActiveNormalCrossing - m_routeman.m_arrival_min) >
            m_routeman.pActivePoint->GetWaypointArrivalRadius()) {
          if (++m_routeman.m_arrival_test > 2 &&
              !g_bAdvanceRouteWaypointOnArrivalOnly)
```

leamas, 2023-05-09:
In my thoughts, still trying find the difference. I can so far see five candidates:

1. Route
2. Navigation data
3. Configuration

We have reasons to believe that none of these is the culprit.

4. Charts
5. Build

Following this scheme: what happens if you test without any charts at all? 

If still no difference: what happens if you test with the build at https://dl.cloudsmith.io/public/alec-leamas/opencpn/raw/names/opencpn%20test/versions/f17877939/opencpn_5.8.0-0+f17877939_setup.exe ?

Hakansv, 2023-05-09:
With a completely clean config file, i.e no charts, the issue is still there.
My build is done by VS2022 - RealWithDebInfo

With the 5.8.0 build as of your link no issue with routes.

leamas, 2023-05-09:
Sorry, I talk too much and read too little:

> With the 5.8.0 build as of your link no issue with routes.

As I see it, this means that this somehow is related to you builds and/or how you run them. Assuming that you also build f17877939  the situation is that this bug occurs for you, but not for me and Pavel. 

I really know too little about the windows development tools  and environment to say anything useful about the possible reasons for this. Anyone, out there?

leamas, 2023-05-09:
> My build is done by VS2022 - RealWithDebInfo

And my with VS2022 - Release

leamas, 2023-05-10:
@Hakansv : I hate chasing you around like this,  please believe me. But I don't see any other way forward as long as I'm not able to reproduce the issue myself.

I assume that you have built this from within Visual Studio. An obvious question is what happens if you build package the same way as me and as I assume Pavel  i. e., by running _ci\\appveyor.bat_ as described in the [manual](https://opencpn-manuals.github.io/main/opencpn-dev/windows.html#_5_8_0_fast_track). This ends up in a setup.exe. Does this exhibit this bug?

Another  data point would be if you could run a `git bisect`  between  5a198001 (good) and 81cfdf93 (bad). That should reveal the exact change triggering the issue for you. Looks like about four iterations, it's not that bad. Knowing the commit might perhaps shed some more light on what's going on here.

Hakansv, 2023-05-10:
I agree about the strangeness. I'll make the tests you suggest this evening.

Connected or not...
This version don't close down as it should but with a crash.
In debug it hangs in nmea0183.cpp:127
```
NMEA0183::~NMEA0183()
{
   initialize();
}
```

leamas, 2023-05-10:
if an issue, this is another one. Let's focus on the original  one here.

transmitterdan, 2023-05-10:
FWIW, I too have tried to duplicate but so far have failed to do so.

leamas, 2023-05-10:
@transmitterdan : It's actually worth a lot. Thanks!

Hakansv, 2023-05-10:
Since I'm the bad example here I restarted from scratch with an empty \build.
Builds fine and made first a prompt -build....   - Works without this issue.
Then built from within VS2022 and debug:
Now no longer a hit here until relevant:
```
if ((m_routeman.CurrentRangeToActiveNormalCrossing - m_routeman.m_arrival_min) >
            m_routeman.pActivePoint->GetWaypointArrivalRadius()) {
```
and all is fine running routes also on my PC as for all others.
Sorry for this false report and I do appreciate all involvement. 
Lessons learned - Clean and rebuild if experiencing a strange bug.
Thanks

closed 2023-05-10 by Hakansv