OpenCPN/OpenCPN #3205: Unable to send routes to GPS over N0183 on version 5.8.2

ka11i0, 2023-05-10:
**Describe the bug**
I updated from 5.6.2 to 5.8.2, and now I cannot send routes to the GPS over NMEA 0183 anymore.
Same issue on Linux and Windows.

**To Reproduce**
Steps to reproduce the behavior:
1. Make a new route.
2. Right click and select "Send to GPS".

**Expected behavior**
Route should be uploaded to the connected GPS unit.

Downgraded to 5.6.2 and it works without any issues.

nohal, 2023-05-10:
What GPS unit?

Hakansv, 2023-05-12:
For info:
O v5.8.2 on W10
Using "Use Garmin GRMN (host) mode..." I can send a route, 3 WPs, to my Garmin "GPS 120 xl" device and it's well received.

ka11i0, 2023-05-22:
It's a custom unit. Sending routes over serial loopback works fine on 5.6.2 but not on 5.8.2.

nohal, 2023-05-22:
I think we still need more info :)

nohal, 2024-02-14:
One more shift in hope the info will be added, I see no other option but to close this unless some device where it actually fails is found and the troubleshooting performed,

leamas, 2024-04-28:
No device seems to be found. Closing, please report in closed bug if an actual device shows the problems described here so it can be reopened.

closed 2024-04-28 by leamas