OpenCPN/OpenCPN #3213: AIS MoB beacon test not showing up as SART or causing alert

antipole2, 2023-05-12:
**Describe the bug**
When an AIS test alert is received from an MoB beacon, it appears as an ordinary AIS target.
Marking the MMSI as MoB seems to have no effect.

**To Reproduce**
Steps to reproduce the behaviour:
1. trigger test alert

**Expected behaviour**
Expect to have attention drawn to the alert and sound played if selected.

**Screenshots**
AIS target list with MMSI 972603802 at top
[image]
AIS target query
[image]
MMSI properties window
[image]
MMSI properties edit window
[image]
VDR log during test
[AIS_test.txt](https://github.com/OpenCPN/OpenCPN/files/11463567/AIS_test.txt)

Also tried marking own ship MMSI as MoB.  No effect.
Does it need to be a real SART (not a test) to be noticed?

**Desktop (please complete the following information if applicable):**
 - OS: MacOS Ventura
 - OCPN 5.8.2

rgleason, 2023-05-13:
The checksums appear to be good. File ais-test.txt does not produce  PLB MOB signals for me on Windows + O5.8.2

Hakansv, 2023-05-13:
> Does it need to be a real SART (not a test) to be noticed?

Yes I think so. The MMSI for a SART starts with 97 and 972 is called a MOB.
OCPN gives an AIS target Alert if the NavStatus = 14 >> "Active".  If undefined NavStatus >> "Testing" no Alert just a AIS target.

antipole2, 2023-05-14:
Am waiting for problems with plugin VDR to be resolved, so I can test this.

There is, perhaps, a case for being able to set a selected MMSI trigger an MoB alert even when it is not a real SART.  This would be useful for test purposes and for crew exercise/drill.  This would be an enhancement.

rgleason, 2023-05-14:
It should work, even if the file window shows the wrong command prompt.
You could use TransmitterDan's VDRplayer   https://github.com/transmitterdan/VDRplayer

rgleason, 2023-05-15:
Windows 11 O5.8.2  The MOB shows next to the AIS for the boat, and the MOB shows in the AIS List, but there is no red distress indication and no sound. 
[image]
[image]
[image]
[AIS_test.txt](https://github.com/OpenCPN/OpenCPN/files/11477421/AIS_test.txt)

Hakansv, 2023-05-15:
The Nav-status for the SART-MOB 972603802 is "undefined" = "Testing" so no Alert as O is programmed. (Fair according to me).
Any SART will activated the AIS-Alert function only when Nav-status = 14, i.e. "Active"

rgleason, 2023-05-16:
Good. Closing

closed 2023-05-16 by rgleason