OpenCPN/OpenCPN #3216: OpenCPN 5.8.2 and legacy installation of its plugins

filochard, 2023-05-12:
Hi
I have been creating rpm packages for OpenCPN and its plugins for years

until 5.6.2 version, Opencpn could find the extra plugins libraries installed besides the bundled plugins libraries (libwmm_pi.so for instance) inside /usr/lib/opencpn/ directory

I know that you promote the flatpak version of OpenCPN for linux distributions so that it can use the Plugin-Manager to install and add plugins

Nevertheless I intended to continue to package OpenCPN 5.8.2 and its plugins to allow an easy update on rpm based linux distributions preserving the previous configurations and data

I succeeded to package OpenCPN 5.8.2 and lots of plugins (radar, ais-radar, watchdog, climatology, weatherfax....) and to install those rpms :
OpenCPN starts smoothly...
But the installed plugins are not in the list of available plugins

Their libraries are correctly installed in /usr/lib/opencpn/ directory besides the bundled plugins libraries (libwmm libchartdldr ...)
But contrary to 5.6.2 version OpenCPN 5.8.2 seems not to look for these plugins there, and they don't appear anymore in the list of plugins to activate...

Is there a workaround to allow this new version to continue to accept legacy installation of plugins ?

If not I will have to stop maintaining packages for linux distributions and to advise OpenCPN users to switch to flatpak 
(which is an heavy way to use OpenCPN : flatpak itself asks for regular updates which needed to download some Go of updates since I test it !!!)

nohal, 2023-05-12:
You may apply this patch during your packaging workflow if you want to allow loading of all the plugins from the system location.

```diff
diff --git a/src/plugin_loader.cpp b/src/plugin_loader.cpp
index a6797ce41..f5acffe42 100644
--- a/src/plugin_loader.cpp
+++ b/src/plugin_loader.cpp
@@ -101,6 +101,8 @@ PlugInContainer::PlugInContainer() {
 
 /** Return true if path "seems" to contain a system plugin */
 static bool IsSystemPlugin(const std::string& path) {
+  /** We are packaging for a system not supported by the plugin manager
+   * and do want to distribute our own packaged plugins
   static const std::vector<std::string> SysPlugins = {
     "chartdldr_pi", "wmm_pi", "dashboard_pi", "grib_pi" };
 
@@ -109,6 +111,8 @@ static bool IsSystemPlugin(const std::string& path) {
     if (lc_path.find(p) != std::string::npos) return true;
   }
   return false;
+  */
+  return true;
 }
 
 SemanticVersion PlugInContainer::GetVersion() {
```

I suppose that you may also encounter problems with the locations where plugins expect to find the various data files they require and might need patching as well...

filochard, 2023-05-12:
Thanks a lot
I'm gonna apply this patch and see what happens for the plugins that have been rewritten for the Plugin Manager

bdbcat, 2023-05-12:
I think we should add a config option, hidden, to support this plugin installation mode.  At least on linux builds, where we have a variety of build and support environments.  Seems trivial.

nohal, 2023-05-12:
I thought about that, but it is a bit less trivial than it seems at first as the installation is performed as root while the config file where those options would have to be written has to be in user's home (also of users that might not even exist during the installation).
Perhaps a command line switch is more suitable for this as the packagers do have control of the `.desktop` files they deploy.
My biggest concern though are the API calls like `GetPluginDataDir` vs. plugin's expectations whether these locations will be writable or not...

bdbcat, 2023-05-12:
I was not thinking to that level of detail.  I imagine simply that a packager could arrange to have plugins in /usr/lib/opencpn/, and O will open them (by config option, or CLI option) as "unmanaged" (i.e. unmanageable in user space).
If the plugin install and link, they should run.  But they cannot be updated or deleted by the Plugin Manager.
Such plugins may indeed require tweaks, but this is on the packager to maintain.
I think it is not a great idea to deploy this as a packaged solution, but caveat emptor.
This is the second such request.

leamas, 2023-05-13:
Loading plugins from /usr/ on Linux re-opens a can of worms which (as Pavel points out) basically boils down to that package managers like rpm and dpkg are not designed to handle installations in user-writable directories. OTOH, our plugin model relies on that user has write permissions. It is not only about installing plugins (we could live without that) but also plugin data.

However, what could be possible is to add "shared plugins" maintained centrally by a package manager to the catalog. That should make it possible for a user to "install" these plugins, basically by copying data from /usr to $HOME instead of downloading it as today. This would represent some middle ground where we could keep the basic model that data is stored in $HOME while still supporting plugin maintained by a package manager. 

I'm currently working on #3103. When completed, this will add imported plugins to the effective catalog. With this in place it should be quite easy to implement also "shared plugins" according to these ideas.

But then again, I'm not convinced that we should actually do this. Just wanted to say that if we should implement plugins installed by package managers, this is IMHO a better path.

bdbcat, 2023-05-13:
@leamas...
Agree with you basic logic for a potential future enhancement, but...
The OP of this thread simply wants a plugin _that he has packaged_ to load from /usr/lib/opencpn,   He would like to do this without patching OCPN, even though the patch as Nohal shows is quite trivial.
Whether it works or not is up to him, since this is a non-standard environment.  Of course, plugins that need private writable data storage will potentially need modification.  Many will not, since they may be using private data locations in e.g.~/.opencpn now. But we leave that detail up to the OP, on an individual basis.
So my thought is a simple CLI option that effectively does what Nohal's patch does, programmatically.  Low risk, possibly meaningful reward for this OP.

leamas, 2023-05-13:
If we add this option there is an obvious risk that we will be "locked" into supporting it. And supporting means keeping a can of worms open.

Packagers  are used to applying patches, and it should not be problem to apply the patch above. However, the big difference is that when a packager applies  a patch she has the full responsibility, and we have no commitment  to support it.  IMHO, this is a big advantage. After all, this is a way certain to break the plugin contract we have, so supporting a command line option is just not consistent.

In other words, I think such an option carries a lot of risk, and we should avoid it.

nohal, 2023-05-13:
For 5.8 the simple fact is that patching either the core app, the plugin or both can't be avoided while creating OS-native packages.

The patch above is the simplest scenario, some plugins (sure some of mine) will need a change in logic to distinguish between assets installed by the package like icons or sample config (that will end up in `/usr/local/share`) and data created at runtime (Which will need to be written under `~/.opencpn` or `~/.local/share/opencpn`).

The ultimate solution that will work universally without making the packagers do any patching of course is to expect this in the APIs and implementing the CLI switch, but that "can't be done" now as the 1.18 API is frozen for O5.8.
That we can't protect the packagers from dependency conflicts is a fact, they will have to live with that forever.
Or we may be lucky at some point and bring the heuristics trying to identify plugin libs actually built against incompatible dependencies to absolute perfection.
Or everybody simply accepts that flatpak works (I say that after maintaining distro specific packages for years, and never going to do it again) and we can do something productive.

bdbcat, 2023-05-13:
I'll make one more pass...
I do not see the can of worms, on linux, at least.  When a plugin is loaded from /usr/lib/opencpn, without metadata, it looks more or less exactly like one of the 4 default plugins.  It has a specific "marker" icon in PIM (gray circle, unmanaged).  Whatever private storage it needs is it's own business. Does not disturb the contract for managed plugins in any way.
As an aside, I find that not many existing plugins store private run-time data in ~/.local/share.  Most use ~/.opencpn/plugins.  So if we have a contract there, we sure do not enforce it.

But I hear the arguments, and yield to the voices of reason and consistency.
Ready to close on OP final.

leamas, 2023-05-13:
Started a discussion #3223 on the more general topic.

filochard, 2023-05-13:
> You may apply this patch during your packaging workflow if you want to allow loading of all the plugins from the system location.
>
> ```diff
> diff --git a/src/plugin_loader.cpp b/src/plugin_loader.cpp [...]

Hi and thanks again
I applied your patch and it works
Now the plugins libraries are found where they have been installed (/usr/lib/opencpn) and can be activated

But as you supposed I get some problems with some of them
weatherfax 1.9.56 is supposed not to be complient with OpenCPN 5.8.2 and is removed from the list
weather-routing 1.14.0.0 induces a crash when I try to enter a boat polar
logbookkonni-1.4.30 induces a crash (I already informed about this with opencpn 5.6.2 when I buid it with wxgtk >3.0 )
etc...

It's far to be simple since everything has been rewritten to be used with the Plugin Manager
Besides this I naturally tried the flatpak version which is OK 

The problem is that there will be no way to simply update a rpm based installation
And switching to flatpak will not be so easy if you want to keep the user's data from the rpm installed OpenCPN 5.6.2
and to use the charts that are on a path which flatpak doesn't have access to

Packaging the plugins will need some other patches (I already needed to use some workarounds to package some of them since
the cmake file has been modified to only create tarballs)

nohal, 2023-05-13:
@filochard wouldn't it perhaps be simpler to create the tarballs for your target platform and provide them normally through the plugin manager?

leamas, 2023-05-13:
> and to use the charts that are on a path which flatpak doesn't have access to

It's perfectly possible to give flatpak access to other directories. No link at hand, the user manual which should be the normal reference seems not accessible. But google should reveal multiple ways to fix this, none of which particularly complicated

leamas, 2023-05-13:
> And switching to flatpak will not be so easy if you want to keep the user's data from the rpm installed OpenCPN 5.6.2

Creating a script which moves data from the native data paths to Flatpak's is  probably not that complicated. And it would be a one  time  effort.

filochard, 2023-05-14:
> @filochard wouldn't it perhaps be simpler to create the tarballs for your target platform and provide them normally through the plugin manager?
It is working for the plugins rpms now that I have had the Nohal's  patch to build OpenCPN

nohal, 2023-05-14:
> It is working for the plugins rpms now that I have had the Nohal's patch to build OpenCPN

Are you sure it is actually working? I certainly do not want to discourage you, but I'm absolutely positive some plugins (at least mine) won't work as expected if you don't patch them as well. Hence my question whether your effort required wouldn't be lower if you simply built the tarballs for the platforms of your interest and submitted the PRs with metadata links to https://github.com/OpenCPN/plugins/ instead of maintaining the RPMs. To me it seem like a better long term approach...

filochard, 2023-05-14:
> It's perfectly possible to give flatpak access to other directories. No link at hand, the user manual which should be the normal reference seems not accessible. But google should reveal multiple [...]

I try to do something user friendly that's why I package OpenCPN: 
 it's not very easy for someone who has been using rpm to install and update OpenCPN since 3.2.2 version to switch to flatpak and have to understand why he has no more access to his charts and to google to know how to modify its command line following his own path to these charts

filochard, 2023-05-14:
> Are you sure it is actually working? I certainly do not want to discourage you, but I'm absolutely positive some plugins (at least mine) won't work as expected if you don't patch them as well. Hence [...]

I tested OpenCPN 5.8.2 built with your patch and lots of plugins work well using the previous ./opencpn/ config and data
The only plugin that is not accepted inside OpenCPN 5.8.2 is weatherfax (neither weatherfax 1.9.56 nor even 1.9.54 that works well with OpenCPN 5.6.2)

nohal, 2023-05-14:
As you wish, I tried...

filochard, 2023-05-14:
> > And switching to flatpak will not be so easy if you want to keep the user's data from the rpm installed OpenCPN 5.6.2
> 
> Creating a script which moves data from the native data paths to Flatpak's is probably not that complicated. And it would be a one time effort.

Indeed, the tree from /home/your username/.opencpn created after an rpm installation 
and the tree from /home/your username/.var/app/org.opencpn.OpenCPN/config/opencpn created by flatpak 
are quite identical
But where to provide this script ? and how to inform the rpm users that they have to switch to flatpak

filochard, 2023-05-14:
> Are you sure it is actually working? I certainly do not want to discourage you, but I'm absolutely positive some plugins (at least mine) won't work as expected if you don't patch them as well. Hence [...]

Does it mean that building tarballs implies they are uploaded to cloudsmith or would it be possible to install them with a simple rpm inside a folder where they would be found by the Plugin Manager (which would need a patch to find them inside the computer instead of the cloud)

filochard, 2023-05-14:
> As you wish, I tried...

Hi Nohal
Don't worry about your plugins installed as rpms inside OpenCPN 5.8.2 they work  : 
 polar_pi (I use the source code from Rick Gleason)
objsearch_pi (I use your source code)
iacfleet_pi (I use your source code)

nohal, 2023-05-14:
Building the tarballs does not mean they have to be uploaded to Cloudsmith, they can be hosted anywhere. The only point obligatory to bring everything together is the metadata repository (https://github.com/OpenCPN/plugins/). If you build the tarballs, simply forget about RPMs, they will not exist for the plugins anymore, users of your RPM packaged OpenCPN will get the plugins exactly as everybody else.
And I still do know that none of my newer plugins will work as expected and the old ones will get updated to use the same principles over time, so your patching required will only get worse, not better.

filochard, 2023-05-14:
Ok thanks for this
I'm gonna dig to understand what to do to build tarballs and where to add them so that the Plugin Manager find them

nohal, 2023-05-14:
For the plugin manager to "find" your tarballs, you simply submit a pull request like https://github.com/OpenCPN/plugins/pull/888
The XML metadata are generated as part of the tarball build, for example if I build the above plugin on my Fedora workstation the generated `Launcher-v1.3.5-fedora-38.xml` looks like this
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!-- deltonator - 2305091044 -->
<plugin version="1">
  <name> Launcher </name>
  <version> v1.3.5 </version>
  <release>  </release>
  <summary> Launcher plugin for OpenCPN </summary>

  <api-version> 1.18 </api-version>
  <open-source> yes </open-source>
  <author> Pavel Kalian </author>
  <source> https://github.com/nohal/launcher_pi </source>
  <info-url> https://opencpn.org/OpenCPN/plugins/launcher.html </info-url>

  <description>   Launcher plugin for OpenCPN
 </description>

  <target>fedora-x86_64</target>
  <target-version>38</target-version>
  <target-arch>x86_64</target-arch>
  <tarball-url> https://dl.cloudsmith.io/public/nohal/launcher_pi-stable/raw/names/Launcher-v1.3.5-fedora-38-tarball/versions/v1.3.5/Launcher-v1.3.5_fedora-38-x86_64.tar.gz </tarball-url>
  <tarball-checksum> 61e8fbe1cb9ba9ab258d1a9af78011244b8dce1fcb7e84c9b065ecff8351763a </tarball-checksum>
</plugin>
```
In ideal case this will simply just work, but I don't know what exactly you are packaging for, so maybe we will need to patch something somewhere to make it so. But ironing out this process IMHO makes more sense than eternally swimming against the current and maintaining the RPM packages for plugins, that simply won't work long term.

filochard, 2023-05-15:
Finally everything works...
I rewrote a patch for weatherfax not to use the bundled audiofile headers but to add the ones from our distro...
the rpms of the plugins are installed, recognised by OpenCPN, activated and work
We can keep the data from 5.6.2

filochard, 2023-12-04:
Hi Rick
I don't know what to answer to you
I'm still building rpms for OpenCPN 5.8.4 and its plugins (thanks to Nohal and his patch https://github.com/OpenCPN/OpenCPN/issues/3216#issuecomment-1546587983)  
For weatherfax I don't use the bundled sources but our own packaged sources for audiofile and portaudio...
(same way for other plugins : I always use our own sources instead of the bundled ones, it's a way to not have conflicts between versions )

Flatpak is a sandbox with its own libraries
I don't know if when you build for Flatpak the Build System uses its own sources as they are looked for in CMakeLists or if it uses
the bundled ones

And I don't know if the bundled sources are up to date, (maybe you may try do update them)

rgleason, 2023-12-05:
Well, the simple question that I have right now is, does the standard flatpak version install and work properly?  And from what you have said, the current version fails to work because it is not using the flatpak libs for sound!  This needs to be fixed.

filochard, 2023-12-05:
I didn't well understand your question
What I can say is that with OpenCPN used with its  flatpak version, I get sound 
(I can choose the sounds and test them inside /toolbar/User Interface/Sounds)

But I can't really test weatherfax with audio since I have no HF radio connected (I'm home in center of France !)
I can only get data from internet and it works for this (with flatpak and with my legacy rpm install)

When I navigate with rented boats, I use my laptop but can't connect it to the inboard radio, and I only use weatherfax with data from internet, or I download gribs from dedicated sites for Mediterranean sea ( such as https://openskiron.org/en/ ) that are perfectly displayed over the Charts by OpenCPN

rgleason, 2023-12-06:
Ok, so provided that weatherfax uses the exact same audio drivers that OpenCPN uses for flatpak, then it should work?  Is that a fair assumption?

rgleason, 2023-12-06:
@leamas Good point! So we need to use the flatpak library for sound that is received.

filochard, 2023-12-07:
> For flatpak: probably so.
> 
> But note that weatherfax does something which not is done anywhere else if I get this correctly: it listens to sound input rather than plays sound. When it comes to such functionality, weatherfax is on it's own. In particular, main OpenCPN certainly does not use sound input for anything.

I'm gonna contact an old Mageia packager (the first to package OpenCPN who then passed on the baton to me) 
He is a sailboat sailor, has great knowledges about ship radio station, he's a radio amateur and packages everything related to radio
He uses  OpenCPN and weatherfax (some years ago he tested it and even thought that our BuildRequire for librtlsdr was useless)
He received faxes in North Sea between England and Nederland and could display them
NB for years we have been building weatherfax with our own packaged audiofile source and not the bundled one...

rgleason, 2023-12-08:
@bdbcat Perhaps weatherfax build could be changed to use the library in opencpn-libs  and get the opencpn-libs updated with that library?  I suppose it depends on timing and release plans.

filochard, 2023-12-08:
> @bdbcat Perhaps weatherfax build could be changed to use the library in opencpn-libs and get the opencpn-libs updated with that library that allows receiving sound so that it can be decoded? I suppose it depends on timing and release plans.

Actually the source of audiofile is only in weatherfax tree 
you may ask to add this source to opencpn-libs 
but if you only get audiofile from its github you will get a 7 years old source, which needs to be patched because of CVE-2018-13440 and CVE-2018-17095 and CVE-2022-24599  (and maybe some others : I only looked for the last patches we had to create)

I don't know how do flatpak or Cloudsmith function  , and if they can bring themselves the needed libraries you use to build this weatherfax-plugin for linux
I don't know how you build your tarballs for Windows, Mac OS...

I don't know in which way you may correct the audiofile source you use
I only know that for security reasons the linux distributions have applied patches to audiofile source
I know that for this reason I prefer not to use bundled libraries which are not always corrected; and I use our own packaged libraries (disabling the bundled ones...) to build packages
Security and consistency are the 2 reasons why I prefer to buid rpms for our distribution

bdbcat, 2023-12-08:
@rgleason 
I cannot see any good reason to add audio receive capability to opencpn-libs.  AFAICT, weatherfax_pi will be the only plugin to actually use this functionality.  So i think it is better to have the affected plugin deal with this function internally.
We have other functions in this category.  Example:  GRIB file decoding.  Only GRIB_pi uses this today, no one else.  So adding GRIB decoding logic to opencpn-libs only makes maintenance more difficult, or at least involves more moving parts.

filochard, 2023-12-09:
@rgleason 
NB This should be an issue for weatherfax_pi itself

For weatherfax_pi the only solution seems to apply some patches to the source of audiofile that is bundled,  if you want not to have security vulnerability
I don't know where your audiofile source comes from, but I think it has not been updated for a long time...

If you want to update this source, I give you the list of the patches we have been using since 2017 in the order they have been  applied on the source from debian (you will have to adapt them for your own source of audiofile, and to cherrypick inside them because apparently some of them have already been applied in the source you use)  : 

here is the list of their names and the order we apply them: (NB the CVE in the name of a patch is part of its name, not a link !)
audiofile-0.3.6-CVE-2015-7747.patch   
01_gcc6.patch                                                 
02_hurd.patch                                                 
05_Always-check-the-number-of-coefficients.patch              
06_Check-for-multiplication-overflow-in-MSADPCM-decodeSam.patch
07_Check-for-multiplication-overflow-in-sfconvert.patch
08_Fix-signature-of-multiplyCheckOverflow.-It-returns-a-b.patch
09_Actually-fail-when-error-occurs-in-parseFormat.patch 
10_Check-for-division-by-zero-in-BlockCodec-runPull.patch
audiofile-0.3.6-CVE-2018-13440.patch
audiofile-0.3.6-CVE-2018-17095.patch

and I attach the URL where you can find and download them
http://svnweb.mageia.org/packages/cauldron/audiofile/current/SOURCES/

rgleason, 2023-12-09:
`I don't know where your audiofile source comes from, but I think it has not been updated for a long time...`

I believe the file being used was found by me many years ago.

rgleason, 2023-12-09:
@bdbcat Thanks. Makes good sense to keep an updated audio in the plugin.

nohal, 2024-02-17:
This needs final closure, one way or another.

closed 2024-02-17 by filochard

filochard, 2024-02-17:
> This needs final closure, one way or another.
Indeed there were two issues here : 
one about legacy install of plugins which got a solution with the patch provided by Nohal https://github.com/OpenCPN/OpenCPN/issues/3216#issuecomment-1546044601  that worked for OpenCPN 5.8.2
This may be closed
(I needed to use other patches for OpenCPN 5.8.4 but that's an other problem solved too)

the second about libaudiofile inside weatherfax I provided an updated source to Rick https://github.com/rgleason/weatherfax_pi/issues/160  and this issue is closed now