OpenCPN/OpenCPN #3237: OCPN no longer starts receiving NMEA data from WiFi if that source appears later

antipole2, 2023-05-14:
**Describe the bug**
If a WiFI source of NMEA data becomes available after OCPN is started or the input stream is enabled, OCPN does not start receiving it.

**To Reproduce**
Steps to reproduce the behaviour:
1. Establish working NMEA input from WiFi source
2. Switch off WiFi source
3. Turn it back on.  Data is not listed in NMEA debug window.
4. Turn off the input stream in Connections and click Apply
5. Turn on the input stream and click Apply
6. NMEA data now shows in NMEA debug window

**Expected behaviour**
OCPN should start receiving data when it becomes available - as it did in OCPN 5.2.

Users often have OCPN running before turning on their navigational system.  OCPN should start processing the data as soon as it becomes available withouthe need to restart OCPN or disable/enable the input.

**Desktop (please complete the following information if applicable):**
 - OS: MacOS Ventura
 - OCPN v5.8.2

nohal, 2023-05-14:
What is "WiFi"? Is it TCP or UDP data stream? If TCP, which side is the server - OpenCPN or the device?

antipole2, 2023-05-14:
OCPN is TCP <--> Minihub Router <--> W2K NMEA 2000-NMEA converter
[image]
The W2K is in what Actisense call 'client mode' i.e. it is not providing the routing.  That is done by the  BT Minihub router.
But I do provide the TCP address of the W2K, which is serving the data. So OCPN is client?

nohal, 2023-05-14:
Yes, OpenCPN is the client in this scenario. And while switching off the network you kill the W2K or the router providing the wi-fi network itself?

antipole2, 2023-05-14:
Yes - when not on passage the W2K and the router are not powered. When about to start passage these are powered up and OCPN used to start handling the data once the computer was connected to the WiFi network.

nohal, 2023-05-14:
In 5.8 we have disabled the logic triggering reconnects when no data was received on a connection (even when the socket was alive) as it was clearly not the right thing to do. But in your case there obviously can't be any open socket if you kill even the network itself and we should try to reconnect periodically. Will have a look at it.

nohal, 2023-05-15:
Can't reproduce whatever I try. Can you try having a look at the network using Wireshark?

This is what I see when I try to reproduce between the Mac (192.168.0.12) and the NMEA TCP server (192.168.0.200:10110)
Looks exactly the same regardless of whether I only don't have the server running or actually disconnect the Mac from the WiFi and then reconnect it back.
[image]

antipole2, 2024-05-09:
Just to confirm, I am still seeing this with OCPN 5.9.0-universal+8e73d3445 (latest I have - no Internet WiFi afloat).

If I switch on boat electronics (instruments and on-board local WiFi) when OCPN already running, no NMEA0183data.
To get data, I need to either

1. restart OCPN or
2. disable the input connection, click Apply, re-enable it and click Apply again.

antipole2, 2024-07-02:
Not fixed (fully?) in Beta 2.
If OCPN loses WiFi connection, data not received after connection restored.  Still aveto disable and re-enable in Connections.
Test by turning WiFi off and,after loss has been registered, turn back on again

antipole2, 2024-07-22:
Still seeing this in v5.9 Beta 3

antipole2, 2024-07-23:
No - it is much improved.  If WIFi appears after instruments sending data, it starts using it.

antipole2, 2024-08-02:
Inconsistent results with Beta 4.
If NMEA0183 data appears in an established WiFi connection, it starts to use it.
When a WiFi connection is made late, it does not seem to start using the data - unless I disable and re-enable the connection.

Hakansv, 2024-08-02:
Confirmed on Win 10 for TCP and SignalK.

bdbcat, 2024-08-02:
All that is certainly to be expected.  There is no cross-platform "notification" mechanism for arriving and departing WiFi connections.  Our alternative would be to run some sort of timer loop in each of the network connection drivers, checking for a new (or lost) connection periodically.
We had something like this in early 5.x releases.  Removed during the major refactor of connection drivers in 584+.
This may be a  reasonable RFE for O5.12.

leamas, 2024-08-02:
There is s starter, we do listen to some dbus events on Linux for serial connections going up and down. But making this work for all platforms is a major undertaking.

Hakansv, 2024-08-02:
On the other hand if wifi TCP is up when O is started and later going down it will be recognized if it later comes up again. So some kind of timer would be involved for an enabled connection.
I though agree this is for v5.12

sebastien-rosset, 2024-12-16:
It would be great to consider #4251 such that:
1. Connect/disconnect events are available as events, not popup windows.
2. OpenCPN retries to connect every X minutes (perhaps with exponential backoff) after disconnect.

bdbcat, 2025-03-14:
Coming back around to this particular Issue, and leaving to one side the question of modal popups:

In current code base, TCP server providing N0183 data, and linux client connecting (as it happens) over a Wifi link, then all works as expected.  For example:
Turn OFF server.
Shutdown OCPN.
Start OCPN with client enabled.  No data, obviously.  Modal dialog appears.  Dismiss dialog.
Turn ON server.
In due course, client begins receiving nav data.

I would like to get verification of this basic functionality on Windows and Mac platforms.

Hakansv, 2025-03-14:
On Win I can reproduce the failure to reconnect.
Following your 1-6 sequence above.
4. Turn ON server
5. No - For a normal* TCP source the client NOT begins to receiving data. 
6. When cycle OFF/ON  client begins receiving nav data.  Now and always after.

*) "Normal" source
An observation:
For a source with data streaming more than 1 Hz the client is not receiving data. But - Another TCP source where there is a pause in the broadcast for several seconds from time to time is always reestablished at the sequence item 5.

Note: I don't think the modal dialog itself has anything to do with this. It's not about a strange relationship between connections logic and GUI logic.

sebastien-rosset, 2025-03-14:
I've tried on macOS sequoia 15.3.1 with OpenCPN master build. I am using a NMEA simulator that send messages to UDP or accept connections over TCP. I tested the following scenarios:
1. NMEA 2000 over TCP where simulator is on `localhost`.
2. NMEA 0183 over TCP where simulator is on `localhost`.

 I can start/stop the simulator many times, and OpenCPN reconnects.

bdbcat, 2025-03-15:
I confirm that this is a real bug, though elusive.
Setup:
0.  Windows 10 laptop, or MacOS laptop.
1. System using Wifi connection to some TCP server providing N0183 data.  No other network connections.
2. Enable the connection in OCPN.  Verify that it works.
3. Stop OCPN.
4. Disable the PC Wifi connection (Airplane mode works easily)
5. Start OCPN.
6. Wait for connection to automatically establish.  The timed retry loops in the code use 8 second timeout.

Results:  For some data streams, the connection is correctly setup on retry, and data flow starts as expected.  For some other data streams, the connection never appears.
On investigation, we find that the _successful_ connection provides a relatively sparse data flow rate,  often with multi-second intervals between packets.  Something like one might see from an AIS receiver in target poor environment.
However, the _unsuccessful_ connection is observed to have a much higher data flow rate.

We tentatively postulate that the trouble is in the wxWidgets sockets implementation for these two platforms, at least.
Rationale:  Two known working connections with identical networks stacks behave differently, depending on their individual data rates.

Confirmation and thoughts appreciated.

Hakansv, 2025-03-16:
For the record if the future leads to more work on this.
There seems to be a inconsistency in the error handling. The "Communication error" message dialog is not created for a Signalk connection. Only for TCP.

Test example: Two TCP and one Signalk connection are enabled. 
So when the wifi is not working on a O start up there are error dialogs for the TCP but not for the Signalk.

When wifi is later restarted the "slow", see above, TCP but also the SignalK connections are established. 
Note that the SignalK connection contains the same data flow as the "fast" TCP that fails to connect!
So if we assume a socket bug making the TCP to not connect why is the same data flow connected when it's SignalK. The SK connection does also use a "TCP" socket, or?

```
These are the error dialog converted to log messages:
09:39:07.326 MESSAGE ocpn_frame.cpp:640 Communication Error Cannot connect to remote server signalk.stupan.se:10114
09:39:07.326 MESSAGE ocpn_frame.cpp:640 Communication Error Cannot connect to remote server edit: "a private slow IP"
09:39:17.095 MESSAGE comm_bridge.cpp:182    ***GPS Watchdog timeout at Lat:57.6452   Lon: 11.7001
Here I restarted the wifi service:
09:39:50.169 MESSAGE comm_drv_signalk_net.cpp:415 Connected to Signal K server version: 2.13.2
09:39:52.478 MESSAGE comm_drv_n0183_net.cpp:447 TCP NetworkDataStream connection established: TCP: edit: The "slow" TCP
```

closed 2025-06-25 by antipole2