OpenCPN/OpenCPN #3243: AIS DSC alerts

rgleason, 2023-05-14:
**Describe the bug**
Hitting the silence button should silence all subsequent alerts, not just that alert.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the file AIS_dsc_mob-distress-simple-gibraltar.txt  attached below
[AIS_dsc_mob-distress-simple-gibraltar.txt](https://github.com/OpenCPN/OpenCPN/files/11473210/AIS_dsc_mob-distress-simple-gibraltar.txt)

2. Options > User > Sounds  AIS Sounds  check play AIS Alert sounds.
3. Use VDR to run the file.
4. A sound alarm plays, but disabling by "Silence Alert" only disables that instance.
5. Also the AIS DSC alert does not show up under the AIS Target List!!!

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version 5.8.2

rgleason, 2023-05-15:
Als
[AIS_dsc-mob-distress-simple-tokyo.txt](https://github.com/OpenCPN/OpenCPN/files/11473238/AIS_dsc-mob-distress-simple-tokyo.txt)
o similar with

Hakansv, 2023-05-15:
Playing a VDR is a test tool and not always a perfect picture of a real situation.

4. The VDR does play by repeat. I.e. when played to end of file it starts again. That's a "new" distress. 
5. The max range in your AIS target list is set to 40 NM. What I can see the distance to own ship is more than that?

rgleason, 2023-05-15:
Yes. So it is working correctly.

closed 2023-05-15 by rgleason