OpenCPN/OpenCPN #3244: AIS_dsc-position-report-test.txt

rgleason, 2023-05-15:
**Describe the bug**
AIS_dsc-position-report-test.txt   do not report and are not seen in the AIS Target List.
They should appear over Japan somewhere.

**To Reproduce**
Steps to reproduce the behavior:
1. Use  AIS_dsc-position-report-test.txt
[AIS_dsc-position-report-test.txt](https://github.com/OpenCPN/OpenCPN/files/11473259/AIS_dsc-position-report-test.txt)

2. Use VDR to play the file.

**Expected behavior**
See this screen shot
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version  5.8.2

Hakansv, 2023-05-15:
What's the distance to own ship and what's the max range to show in AIS target list?

rgleason, 2023-05-15:
Changed max range to 20000nm and found these located in Australia, Sydney Harbor.
They are working.
[image]

closed 2023-05-15 by rgleason