OpenCPN/OpenCPN #3245: AIS_mob-buddy-log-southhampton-bramble-aella-13m.txt

rgleason, 2023-05-15:
**Describe the bug**
AIS MOB Buddy  does not show and does not appear at the top of the AIS List.

**To Reproduce**
Steps to reproduce the behavior:
1. Use  
[Uploading AIS_mob-buddy-log-southhampton-bramble-aella-13m.txt…]()

2. Play it with VDR_pi

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version 5.8.2

Hakansv, 2023-05-15:
The link is to this page and not to a file

rgleason, 2023-05-15:
Sorry, fixed.

rgleason, 2023-05-15:
Don't see any MOB, perhaps it needs to be entered in the MMSI spreadsheet? Whatever the mmsi # is....?

Hakansv, 2023-05-15:
The DSC in that file is this (and another similar):
$CDDSC,20,2359129290,00,21,26,1505100118,1421,,,B,E*7F
First field is 20 = " a selective call to a particular individual station" (ITU-R M.493-10)  so no distress.

closed 2023-05-16 by rgleason