OpenCPN/OpenCPN #3246: AIS_sart-alert-test-guernesey-3m.txt

rgleason, 2023-05-15:
**Describe the bug**
Why is this test file working when the others do not?
It also is silenced for once and all, when you hit silence alert.
This is not a bug, but an example. 
Why is it working?

**To Reproduce**
Steps to reproduce the behavior:
1. Use this file 
[AIS_sart-alert-test-guernesey-3m.txt](https://github.com/OpenCPN/OpenCPN/files/11473283/AIS_sart-alert-test-guernesey-3m.txt)
2. Use VDR to play this file.

**Expected behavior**
A clear and concise description of what you expected to happen.
Why is this test file working when the others do not?
It also is silenced for once and all, when you hit silence alert.
It actually works this way.

However I think it should show in the AIS List and be at the top as SART  and mmsi 97000099

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: Windows 11
 - Version 5.8.2

Hakansv, 2023-05-15:
When things are working it would not be called a bug I suppose? Discussion?

> However I think it should show in the AIS List and be at the top as SART and mmsi 97000099

What's the distance to own ship? More than stated max range 40 NM?

rgleason, 2023-05-15:
Yes more than Max distance. Thanks.
These are examples that work, which may show that the example files that are "bugs" may not be good nmea0183, or perhaps may not be supported PLB MOB?

closed 2023-05-15 by rgleason